FS-Driverless/Formula-Student-Driverless-Simulator #286: UE4 crashes after building AirSim and Blocks

MattBrth, 2022-03-03:
Hello,

I managed to clone build and finally start FSOnline.uproject but after rebuilding AirSim and Blocks, it craches at 71% of the loeading screen.

If anyone knows how to fix this that would be great.
 
I got this log : 

---------------------------------------------------------------------

Not using cached read wrapper
Started task graph with 5 named threads and 59 total threads with 3 sets of task threads.
Stats thread started at 0.163176
Loaded GFSDK_Aftermath_Lib.x64.dll
ICU TimeZone Detection - Raw Offset: +1:00, Platform Override: ''
Mounting plugin Paper2D
Mounting plugin AISupport
Mounting plugin LightPropagationVolume
Mounting plugin CameraShakePreviewer
Mounting plugin ActorLayerUtilities
Mounting plugin AnimationSharing
Mounting plugin CLionSourceCodeAccess
Mounting plugin CodeLiteSourceCodeAccess
Mounting plugin GitSourceControl
Mounting plugin KDevelopSourceCodeAccess
Mounting plugin NullSourceCodeAccess
Mounting plugin PerforceSourceControl
Mounting plugin PlasticSourceControl
Mounting plugin RiderSourceCodeAccess
Mounting plugin SubversionSourceControl
Mounting plugin UObjectPlugin
Mounting plugin VisualStudioCodeSourceCodeAccess
Mounting plugin VisualStudioSourceCodeAccess
Mounting plugin XCodeSourceCodeAccess
Mounting plugin AssetManagerEditor
Mounting plugin CryptoKeys
Mounting plugin CurveEditorTools
Mounting plugin DataValidation
Mounting plugin EditorScriptingUtilities
Mounting plugin FacialAnimation
Mounting plugin GameplayTagsEditor
Mounting plugin GeometryMode
Mounting plugin MacGraphicsSwitching
Mounting plugin MaterialAnalyzer
Mounting plugin MobileLauncherProfileWizard
Mounting plugin PluginBrowser
Mounting plugin SpeedTreeImporter
Mounting plugin DatasmithContent
Mounting plugin VariantManagerContent
Mounting plugin AlembicImporter
Mounting plugin AutomationUtils
Mounting plugin BackChannel
Mounting plugin ChaosCloth
Mounting plugin ChaosClothEditor
Mounting plugin ChaosEditor
Mounting plugin ChaosNiagara
Mounting plugin ChaosSolverPlugin
Mounting plugin CharacterAI
Mounting plugin GeometryCache
Mounting plugin GeometryCollectionPlugin
Mounting plugin GeometryProcessing
Mounting plugin PlanarCut
Mounting plugin PlatformCrypto
Mounting plugin ProxyLODPlugin
Mounting plugin SkeletalReduction
Mounting plugin Niagara
Mounting plugin MagicLeap
Mounting plugin MagicLeapLightEstimation
Mounting plugin MagicLeapMedia
Mounting plugin MagicLeapPassableWorld
[... 878 lines cut ...]
No distributed automatic mesh merging module available
No distributed automatic mesh merging module available
FSOnline 1.0.0, NetCL: 13144385, EngineNetVer: 14, GameNetVer: 0 (Checksum: 1557323121)
UAISubsystem::UAISubsystem AISubsystem (0000027BC046B280), frame # 0
UAISubsystem::UAISubsystem AIPerceptionSystem (0000027BC045ED40), frame # 0
UAISubsystem::UAISubsystem EnvQueryManager (0000027BC04B5200), frame # 0
Processed 18 localized package path(s) for 1 prioritized culture(s) in 0.015797 seconds
15756 objects as part of root set at end of initial load.
3925280 out of 0 bytes used by permanent object pool.
CloseDisregardForGC: 0/0 objects in disregard for GC pool
Initializing TcpMessaging bridge
Initializing bridge on interface 0.0.0.0:0 to multicast group 230.0.0.1:6666.
The file 'C:/Users/berth/Formula-Student-Driverless-Simulator/UE4Project/Plugins/AirSim/Content/Blueprints/AirsimSpectatorPawn.uasset' contains unrecognizable data, check that it is of the expected type.
The file 'C:/Users/berth/Formula-Student-Driverless-Simulator/UE4Project/Plugins/AirSim/Content/Blueprints/AirsimSpectatorPawn.uasset' contains unrecognizable data, check that it is of the expected type.
Failed to find object 'Class /AirSim/Blueprints/AirsimSpectatorPawn.AirsimSpectatorPawn_C'
CDO Constructor (AirSimGameMode): Failed to find /AirSim/Blueprints/AirsimSpectatorPawn.AirsimSpectatorPawn_C

The file 'C:/Users/berth/Formula-Student-Driverless-Simulator/UE4Project/Plugins/AirSim/Content/Blueprints/BP_PIPCamera.uasset' contains unrecognizable data, check that it is of the expected type.
The file 'C:/Users/berth/Formula-Student-Driverless-Simulator/UE4Project/Plugins/AirSim/Content/Blueprints/BP_PIPCamera.uasset' contains unrecognizable data, check that it is of the expected type.
Failed to find object 'Class /AirSim/Blueprints/BP_PIPCamera.BP_PIPCamera_C'
CDO Constructor (CarPawn): Failed to find /AirSim/Blueprints/BP_PIPCamera.BP_PIPCamera_C

The file 'C:/Users/berth/Formula-Student-Driverless-Simulator/UE4Project/Plugins/AirSim/Content/VehicleAdv/WheelData/Slippery.uasset' contains unrecognizable data, check that it is of the expected type.
The file 'C:/Users/berth/Formula-Student-Driverless-Simulator/UE4Project/Plugins/AirSim/Content/VehicleAdv/WheelData/Slippery.uasset' contains unrecognizable data, check that it is of the expected type.
Failed to find object 'Object /AirSim/VehicleAdv/WheelData/Slippery.Slippery'

wouter-heerwegh, 2022-03-03:
Hi @MattBrth,

First of all, are you just trying to run the simulator? If so, it's easier to use the compiled binaries. They can be found [here](https://github.com/FS-Driverless/Formula-Student-Driverless-Simulator/releases). When running these, nake sure that the settings.json can still be found in `C:\Users\berth\Formula-Student-Driverless-Simulator`

Otherwise, could you check if the file `Slippery.uasset` (see your log) contains any data or just a git-lfs hash?

If it contains a git-lfs hash, install git-lfs and `git lfs pull` the large files in the repository

MattBrth, 2022-03-03:
No I don't just need the compiled version. 
The `Slippery.uasset` effectively contains a git lfs tag but I dont know how to execute the `git lfs pull` command as I downloaded the zip file of the 2.1.0 version and did not clone the unrealeased repository.
Thank you for your help.

wouter-heerwegh, 2022-03-03:
Hi @MattBrth,

I see, unfortunately we can't commit the files regularly into the repo, because github will not commit files larger than 50MB. As a solution, these are stored on git lfs.

You could do 2 things, remove your folder and clone the repo. You can then checkout on the v2.1.0 tag and git lfs pull. 

Or you could clone the repo in a different folder run git lfs pull and replace the content folder of UE4Project. But I'm not sure if that folder is the obly one containing these large files.

wouter-heerwegh, 2022-03-09:
@MattBrth were you able to build everything with the LFS files pulled?

MattBrth, 2022-03-09:
I was not able to pull a released version but by cloning the master branch I managed to make it work after some time ! 
Thank you for your help.

closed 2022-03-09 by MattBrth