FS-Driverless/Formula-Student-Driverless-Simulator #289: Can't run updated version of FSDS

loremacchia, 2022-03-17:
Hi, we're using an old version of FSDS on Uneral 4.25.4 and it works great. I tried to git clone the latest version but Unreal asks me to rebuild blocks and AirSim because they were built with a different engine version. If I click "no" it just quits saying the modules are missing or incompatible, and if I click "yes" it says they have to be built manually. I did everything the docs say(run setup and build for AirSim etc.) but the problem persists.

wouter-heerwegh, 2022-03-18:
Hi @loremacchia,

Did you run the setup/build scripts for Airsim? If I don't do this, I get the same problem.
This is also mentioned in the documentation [here in 2.](https://fs-driverless.github.io/Formula-Student-Driverless-Simulator/v2.1.0/getting-started/#installation)

Also, are you trying to do this on linux or windows?

loremacchia, 2022-03-18:
Yes, I did run setup.sh and build.sh, as you can see from this picture: [image]
But the error persists. I'm trying to run the simulator on Ubuntu 20.04 LTS and the old version was, and still is, working fine.
I downloaded the version I'm trying to use from the master branch yesterday, so it should be the latest one. The old one is from February 2021 if that's of any help.

wouter-heerwegh, 2022-03-21:
Hmmm, never had this error before. Do you get any errors in your terminal that is launching Unreal Engine?

loremacchia, 2022-03-30:
No, it seems to be the normal startup sequence that I get when launching the old version.

wouter-heerwegh, 2022-04-01:
I'm currently not able to reproduce your issue, it always seems to launch correctly for me. Maybe it would be nice to have some call, so we can have a closer look together.

mateusz-lichota, 2022-08-26:
For the future, what may help is actually building those modules manually. Below is the command for Linux, and I assume it's very similar for windows:
`bash "<path to Unreal Engine>/Engine/Build/BatchFiles/Linux/RunMono.sh" <path to Unreal Engine>/Engine/Binaries/DotNET/UnrealBuildTool.exe Development Linux -Project="<path to repo>/UE4Project/FSOnline.uproject" -TargetType=Editor -Progress`

I've encountered this issue when updating Unreal Engine from 4.25 to 4.27

mateusz-lichota, 2022-08-26:
@loremacchia do you remember if the error you got was `Engine modules are out of date, and cannot be compiled while the engine is running. Please build through your IDE`?

federicochiesa, 2022-09-26:
> @loremacchia do you remember if the error you got was `Engine modules are out of date, and cannot be compiled while the engine is running. Please build through your IDE`?

Yes, we got that message exactly.

RoyAmoyal, 2024-10-16:
Any solution for that?