FS-Driverless/Formula-Student-Driverless-Simulator #293: Client Connection Error

LucaCane, 2022-04-10:
Hello,

I am trying to run this simulator on Win10 using the Python Client. Following the getting started guide I managed to obtain a driveable car running from release binaries andI have installed the requirements for the Python client.

When I run the python example in the guide or the autonomous.py, I get a "Connection Error" (within the client.confirmConnection) which apparently changes error code at almost every run, or a "Typer Error".
Has anyone already solved this problem?

Thank you

wouter-heerwegh, 2022-04-10:
Hi @LucaCane,

Are you running python in a regular command prompt or in wsl or something?

What you could do to see wheter or not the simulator is detected in whatever environment you're using is by checking if a port (41451) is opened on 127.0.0.1. If not, you mostlikely will have to pass an IP address.

LucaCane, 2022-04-11:
Hi @wouter-heerwegh,
I am running python in a regular command prompt.
Tried to check the port as you said but cmd says that it can't open a connection on that port.

wouter-heerwegh, 2022-04-12:
You could use telnet on windows, telnet 127.0.0.1 41451 or any other port that you used for the simulator. If this doesn't return a blank screen but some connection error, mostlikely you will have to pass your IP to your fsds client as an argument. If this doesn't change anything, try again with your firewall turned off, or add a rule in your firewall to allow the simulator to open the port.

LucaCane, 2022-04-12:
Tried without the firewall but I am unable to connect to such port even with a different IP.

wouter-heerwegh, 2022-04-13:
Could you run the executable in a command prompt? Maybe this way we can get some error information.

LucaCane, 2022-04-13:
Runned the example "autonomous.py" (already present in python folder) with and without passing the IP address to client(). I am posting only one screen since I noted that the message is the same in both cases (It also has changed from something like "WARNING:tornado.general:Connect error on fd 2352: WSAECONNREFUSED" to this [number was different]).

[image]

wouter-heerwegh, 2022-04-13:
This error seems a bit weird, could you try to run the FSDS.exe file in cmd? I'm wondering if this throws any errors.

The current error you're having now seems to be an issue with how you're currently using the socket library. Have you changed anything significantly?

LucaCane, 2022-04-14:
Tried to run FSDS.exe in cmd and it worked without any error.

No i did not changed anything, I only have copied the folder within my user folder and I have extracted the fsds.exe folder.