FS-Driverless/Formula-Student-Driverless-Simulator #294: Error when launch the linux release

chentairan, 2022-05-01:
Hi @wouter-heerwegh 

When I try to start the simulator via fsds-v2.1.0-linux, it shows that the path `/home/wouter/UnrealEngine/Engine/Source/Runtime/VulkanRHI/Private/VulkanDevice.cpp` is not found. I wonder if it is caused by using your absolute path in the process of packing into binary?

[FSOnline.log](https://github.com/FS-Driverless/Formula-Student-Driverless-Simulator/files/8598221/FSOnline.log)

Thanks,
Tairan Chen

wouter-heerwegh, 2022-05-01:
Hi @chentairan,

It's weird that this happens. Could you check v2.0.0 if this also happens?

Ehat is weird to me however is that even though it has my absolute path, it is evaluating this file and having an error on an assertion check `VendorId != EGpuVendorId::Unknown`. It seems to me that it cannot determine what the vendor is of your integrated cpu.

I have a feeling it will be because of the llvmpipe. This has a vendor id 0x10005, and this is not part of any of the vendors listed in vulkan.
```enum class EGpuVendorId { 	Unknown		= -1, 	NotQueried	= 0, 	Amd			= 0x1002, 	ImgTec		= 0x1010, 	Nvidia		= 0x10DE, 	Arm			= 0x13B5, 	Qualcomm	= 0x5143, 	Intel		= 0x8086, };```

Do you use a GPU? If so, it doesn't seem to show up for vulkan for some reason.

chentairan, 2022-05-01:
My GPU is RTX 3060 with Driver Version: 465.19.01, CUDA Version: 11.3

chentairan, 2022-05-01:
I reinstalled the GPU driver and compiled FSOnline from source. I think it solved my problem, thanks a lot!

closed 2022-05-01 by chentairan