FS-Driverless/Formula-Student-Driverless-Simulator #296: Fail to build fsds_ros2_bridge

chentairan, 2022-07-13:
Here is the error logs:

In file included from /opt/ros/humble/include/rclcpp/rclcpp/wait_set.hpp:25,
                 from /opt/ros/humble/include/rclcpp/rclcpp/rclcpp.hpp:168,
                 from /home/chentairan/Formula-Student-Driverless-Simulator/ros2/src/fsds_ros2_bridge/include/airsim_ros_wrapper.h:1,
                 from /home/chentairan/Formula-Student-Driverless-Simulator/ros2/src/fsds_ros2_bridge/src/airsim_ros_wrapper.cpp:1:
/opt/ros/humble/include/rclcpp/rclcpp/wait_set_policies/dynamic_storage.hpp:226:7: note: candidate: ‘rclcpp::wait_set_policies::DynamicStorage::storage_has_entity<rclcpp::ServiceBase, std::vector<std::weak_ptr<rclcpp::ServiceBase> > >(const rclcpp::ServiceBase&, const std::vector<std::weak_ptr<rclcpp::ServiceBase> >&)::<lambda(const int&)>’
  226 |       [&entity](const auto & inner) {return &entity == inner.lock().get();});
      |       ^
/opt/ros/humble/include/rclcpp/rclcpp/wait_set_policies/dynamic_storage.hpp:226:7: note:   no known conversion for argument 1 from ‘const std::weak_ptr<rclcpp::ServiceBase>’ to ‘const int&’
In file included from /usr/include/c++/11/bits/stl_algobase.h:71,
                 from /usr/include/c++/11/memory:63,
                 from /opt/ros/humble/include/rclcpp/rclcpp/rclcpp.hpp:153,
                 from /home/chentairan/Formula-Student-Driverless-Simulator/ros2/src/fsds_ros2_bridge/include/airsim_ros_wrapper.h:1,
                 from /home/chentairan/Formula-Student-Driverless-Simulator/ros2/src/fsds_ros2_bridge/src/airsim_ros_wrapper.cpp:1:
/usr/include/c++/11/bits/predefined_ops.h: In instantiation of ‘bool __gnu_cxx::__ops::_Iter_pred<_Predicate>::operator()(_Iterator) [with _Iterator = __gnu_cxx::__normal_iterator<const rclcpp::wait_set_policies::DynamicStorage::WeakWaitableEntry*, std::vector<rclcpp::wait_set_policies::DynamicStorage::WeakWaitableEntry> >; _Predicate = rclcpp::wait_set_policies::DynamicStorage::storage_has_entity<rclcpp::Waitable, std::vector<rclcpp::wait_set_policies::DynamicStorage::WeakWaitableEntry> >(const rclcpp::Waitable&, const std::vector<rclcpp::wait_set_policies::DynamicStorage::WeakWaitableEntry>&)::<lambda(const int&)>]’:
/usr/include/c++/11/bits/stl_algobase.h:2069:14:   required from ‘_RandomAccessIterator std::__find_if(_RandomAccessIterator, _RandomAccessIterator, _Predicate, std::random_access_iterator_tag) [with _RandomAccessIterator = __gnu_cxx::__normal_iterator<const rclcpp::wait_set_policies::DynamicStorage::WeakWaitableEntry*, std::vector<rclcpp::wait_set_policies::DynamicStorage::WeakWaitableEntry> >; _Predicate = __gnu_cxx::__ops::_Iter_pred<rclcpp::wait_set_policies::DynamicStorage::storage_has_entity<rclcpp::Waitable, std::vector<rclcpp::wait_set_policies::DynamicStorage::WeakWaitableEntry> >(const rclcpp::Waitable&, const std::vector<rclcpp::wait_set_policies::DynamicStorage::WeakWaitableEntry>&)::<lambda(const int&)> >]’
/usr/include/c++/11/bits/stl_algobase.h:2114:23:   required from ‘_Iterator std::__find_if(_Iterator, _Iterator, _Predicate) [with _Iterator = __gnu_cxx::__normal_iterator<const rclcpp::wait_set_policies::DynamicStorage::WeakWaitableEntry*, std::vector<rclcpp::wait_set_policies::DynamicStorage::WeakWaitableEntry> >; _Predicate = __gnu_cxx::__ops::_Iter_pred<rclcpp::wait_set_policies::DynamicStorage::storage_has_entity<rclcpp::Waitable, std::vector<rclcpp::wait_set_policies::DynamicStorage::WeakWaitableEntry> >(const rclcpp::Waitable&, const std::vector<rclcpp::wait_set_policies::DynamicStorage::WeakWaitableEntry>&)::<lambda(const int&)> >]’
/usr/include/c++/11/bits/stl_algo.h:3910:28:   required from ‘_IIter std::find_if(_IIter, _IIter, _Predicate) [with _IIter = __gnu_cxx::__normal_iterator<const rclcpp::wait_set_policies::DynamicStorage::WeakWaitableEntry*, std::vector<rclcpp::wait_set_policies::DynamicStorage::WeakWaitableEntry> >; _Predicate = rclcpp::wait_set_policies::DynamicStorage::storage_has_entity<rclcpp::Waitable, std::vector<rclcpp::wait_set_policies::DynamicStorage::WeakWaitableEntry> >(const rclcpp::Waitable&, const std::vector<rclcpp::wait_set_policies::DynamicStorage::WeakWaitableEntry>&)::<lambda(const int&)>]’
/usr/include/c++/11/bits/stl_algo.h:471:47:   required from ‘bool std::none_of(_IIter, _IIter, _Predicate) [with _IIter = __gnu_cxx::__normal_iterator<const rclcpp::wait_set_policies::DynamicStorage::WeakWaitableEntry*, std::vector<rclcpp::wait_set_policies::DynamicStorage::WeakWaitableEntry> >; _Predicate = rclcpp::wait_set_policies::DynamicStorage::storage_has_entity<rclcpp::Waitable, std::vector<rclcpp::wait_set_policies::DynamicStorage::WeakWaitableEntry> >(const rclcpp::Waitable&, const std::vector<rclcpp::wait_set_policies::DynamicStorage::WeakWaitableEntry>&)::<lambda(const int&)>]’
/usr/include/c++/11/bits/stl_algo.h:490:27:   required from ‘bool std::any_of(_IIter, _IIter, _Predicate) [with _IIter = __gnu_cxx::__normal_iterator<const rclcpp::wait_set_policies::DynamicStorage::WeakWaitableEntry*, std::vector<rclcpp::wait_set_policies::DynamicStorage::WeakWaitableEntry> >; _Predicate = rclcpp::wait_set_policies::DynamicStorage::storage_has_entity<rclcpp::Waitable, std::vector<rclcpp::wait_set_policies::DynamicStorage::WeakWaitableEntry> >(const rclcpp::Waitable&, const std::vector<rclcpp::wait_set_policies::DynamicStorage::WeakWaitableEntry>&)::<lambda(const int&)>]’
/opt/ros/humble/include/rclcpp/rclcpp/wait_set_policies/dynamic_storage.hpp:223:23:   required from ‘static bool rclcpp::wait_set_policies::DynamicStorage::storage_has_entity(const EntityT&, const SequenceOfEntitiesT&) [with EntityT = rclcpp::Waitable; SequenceOfEntitiesT = std::vector<rclcpp::wait_set_policies::DynamicStorage::WeakWaitableEntry>]’
/opt/ros/humble/include/rclcpp/rclcpp/wait_set_policies/dynamic_storage.hpp:351:33:   required from here
/usr/include/c++/11/bits/predefined_ops.h:318:30: error: no match for call to ‘(rclcpp::wait_set_policies::DynamicStorage::storage_has_entity<rclcpp::Waitable, std::vector<rclcpp::wait_set_policies::DynamicStorage::WeakWaitableEntry> >(const rclcpp::Waitable&, const std::vector<rclcpp::wait_set_policies::DynamicStorage::WeakWaitableEntry>&)::<lambda(const int&)>) (const rclcpp::wait_set_policies::DynamicStorage::WeakWaitableEntry&)’
  318 |         { return bool(_M_pred(*__it)); }
      |                       ~~~~~~~^~~~~~~
In file included from /opt/ros/humble/include/rclcpp/rclcpp/wait_set.hpp:25,
                 from /opt/ros/humble/include/rclcpp/rclcpp/rclcpp.hpp:168,
                 from /home/chentairan/Formula-Student-Driverless-Simulator/ros2/src/fsds_ros2_bridge/include/airsim_ros_wrapper.h:1,
                 from /home/chentairan/Formula-Student-Driverless-Simulator/ros2/src/fsds_ros2_bridge/src/airsim_ros_wrapper.cpp:1:
/opt/ros/humble/include/rclcpp/rclcpp/wait_set_policies/dynamic_storage.hpp:226:7: note: candidate: ‘rclcpp::wait_set_policies::DynamicStorage::storage_has_entity<rclcpp::Waitable, std::vector<rclcpp::wait_set_policies::DynamicStorage::WeakWaitableEntry> >(const rclcpp::Waitable&, const std::vector<rclcpp::wait_set_policies::DynamicStorage::WeakWaitableEntry>&)::<lambda(const int&)>’
  226 |       [&entity](const auto & inner) {return &entity == inner.lock().get();});
      |       ^
/opt/ros/humble/include/rclcpp/rclcpp/wait_set_policies/dynamic_storage.hpp:226:7: note:   no known conversion for argument 1 from ‘const rclcpp::wait_set_policies::DynamicStorage::WeakWaitableEntry’ to ‘const int&’
gmake[2]: *** [CMakeFiles/airsim_ros.dir/build.make:76: CMakeFiles/airsim_ros.dir/src/airsim_ros_wrapper.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:192: CMakeFiles/airsim_ros.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< fsds_ros2_bridge [4.94s, exited with code 2]

mateusz-lichota, 2022-07-19:
You're trying to build the ros2 bridge for ROS Humble, but the version it was written for is galactic, so you might be a pioneer of the new version here :+1: . If you figure out what the issue is, please do submit a pull request fixing it or at least share the steps to fix it here

b1n-ch1kn, 2022-08-28:
We're starting to look at migrating given Galactic is EOL in November, and Humble has 5 years left.
Was able to build for Humble, need to do some more testing (mostly on our driverless stack) but these changes shouldn't break Galactic. https://github.com/QUT-Motorsport/Formula-Student-Driverless-Simulator/commit/3f3b44782d1fcb2267ed4ab85842406e68775f62

Note, we use RoboStack for ROS+Python+CPP package management. Might need more adjustments for a deb install of Humble.

mateusz-lichota, 2022-09-03:
Closed in #317

closed 2022-09-03 by mateusz-lichota

b1n-ch1kn, 2022-09-04:
Probably moreso fixed here https://github.com/FS-Driverless/Formula-Student-Driverless-Simulator/compare/master...AdMorgado:Formula-Student-Driverless-Simulator:master. Can just change the airsim requirements to newer clang

mateusz-lichota, 2022-09-04:
Wait, I don't have access to Humble so I just trusted you @b1n-ch1kn that your PR fixed this. Does it successfully build on Humble now, or not?

b1n-ch1kn, 2022-09-04:
Sorry for the confusion, my PR was for 20.04 Humble & Galactic compatibility. However, I did start some testing on a 22.04 environment last night after discussing with AdMorgado. 
Seems like the only difference between Humble 22.04 and Galactic 20.04 is clang 14 vs 8 specified in `AirSim/setup.sh` and `AirSim/build.sh`. Unfortunately, 20.04 doesn't have clang 14 support so the same file wouldn't have backward compatibility, but maybe a `setup_2204.sh` or something similar would be the way to go for multi-distro support. 
Happy to hear your thoughts on this so we can support more of the community.

mateusz-lichota, 2022-09-04:
How about instead of clang-14 we try using clang-11 or clang-12, both  of which support both 20.04 and 22.04? Could you see if either one of those versions works with ubuntu 22.04? I'll test 20.04

b1n-ch1kn, 2022-09-04:
Sounds good, I'll try both on the 22.04 end.

mateusz-lichota, 2022-09-04:
Let's go with clang 12 then

b1n-ch1kn, 2022-09-04:
I'll make a PR once I do some more testing tonight. Cheers