FS-Driverless/Formula-Student-Driverless-Simulator #306: Importing custom vehicle into FSDS

b1n-ch1kn, 2022-08-23:
Hi guys, sorry to bother you.
I've been trying to get my team's car into the sim. I've followed the updated tutorial from @mateusz-lichota and his [learnings from importing the ADS vehicle](https://github.com/FS-Driverless/Formula-Student-Driverless-Simulator/blob/master/docs/import-car-3d-model.md).

Unfortunately, I'm stuck with my vehicle not moving when commanded. The animation works and I get command feedback as seen here, but there's no motion from the car:
[image]

It seems like the wheel bounding boxes just aren't colliding with the ground in addition to falling off their joints (the hub screw shoud be centred on the wheels)
[image]

This was how the wheel bodies were set up, and it looks like only the chassis bounding box was hitting the ground, given the depth of the wheels in the ground
[image]

I can't work out if its an UE  issue or blender. But I have defined the meshes in blender according to the tutorial.
[image]

If you wanted to have a play (and you guys have more experience on this than I do) it's on this branch of our fork. If you need our blender or part files, let me know.
https://github.com/QUT-Motorsport/Formula-Student-Driverless-Simulator/tree/merge-upstream
[image]

Thanks!

mateusz-lichota, 2022-08-25:
Hi there! I've tried out your fork, and while I don't really know why the car won't move I've noticed a potential issue: the axes of your vehicle are different than of those which work. Maybe you didn't select the appropriate coordinate system when exporting an .fbx from blender? (It should be x forward). Either way, try recreating and reimporting the .fbx file so that the axes of the chassis and the wheels match the orientation of those on AdsDV and the TechnionCar

b1n-ch1kn, 2022-08-25:
Thanks for having a look! I had been battling the coord frames for some time in both blender and unreal but I'll check out what the other cars should be and try to replicate. Will let you know of progress

b1n-ch1kn, 2022-09-05:
I'll sink some more time into this later, planning for the Aussie comp is taking priority these days.

closed 2022-09-05 by b1n-ch1kn

b1n-ch1kn, 2022-09-05:
That would be awesome, thanks @mateusz-lichota. Sent you Gdrive a link on LinkedIn