FS-Driverless/Formula-Student-Driverless-Simulator #307: Switching to Unreal Engine 4.27

mateusz-lichota, 2022-08-25:
UE 4.27 has a myriad of small improvements over 4.25, and the only change needed to switch is one line of code. This would also set up ground to potentially switch from the old PhysX vehicle api to the new ChaosVehicle api in the future

closed 2022-08-26 by wouter-heerwegh

b1n-ch1kn, 2022-08-29:
One thing I hit when building a new Windows exe for UE4.27 was the custom map loader blueprint giving a 'hotreload' error. Searched it up, meant it needed to be recompiled. But simply clicking recompile in the blueprint didn't fix it, so removing a wire, re-adding, the recompiling fixed it. Noting for anyone in the future who runs into this.

mateusz-lichota, 2022-08-30:
@b1n-ch1kn did fixing this cause a change in the project files as reported by git? If so, could you just commit that change on a new branch and make a PR?

PR #310 (2022-08-26): Switch to Unreal Engine 4.27
This closes #307