FS-Driverless/Formula-Student-Driverless-Simulator #337: Unable to build Airsim plugin in Unreal engine 4.27 with clang++12

wouter-heerwegh, 2022-09-25:
I've been struggling trying to build the latest version on my linux machine (Ubuntu 20.04). I've found out that UE4 uses clang++11, [as seen here](https://docs.unrealengine.com/4.27/en-US/SharingAndReleasing/Linux/GettingStarted/) and I've not been able to force it to use clang from the system.

@b1n-ch1kn any idea?

b1n-ch1kn, 2022-09-25:
Hmm, I'm not too familiar with the docker containers for unreal, and I think the guys on my team with Linux machines use an Epic Games UE base container to run the sim.
Have had a look, as far as I can tell 2204/Humble is compatible with Clang 11 and builds fine, if you can get it to work on that?

wouter-heerwegh, 2022-09-26:
I'm not using a docker container, UE4 is just installed on my system. I don't exactly know what I did, but I managed to build it with clang++-12. Gonna check later on another system to see if I have the same difficulties.

closed 2022-09-26 by wouter-heerwegh