FS-Driverless/Formula-Student-Driverless-Simulator #338: Getting a fatal error when running from pre-compiled binaries

pyszkaj, 2022-09-26:
Hi everyone, 

I'm trying to run the simulator from pre-compiled binaries but as soon as I start the simulator throws a fatal error. I have a fairly high spec PC with vulkan supported GPU. What could be causing this issue?

Thank you in advance

wouter-heerwegh, 2022-09-26:
Hi @pyszkaj 

What OS are you running on (Windows, Ubuntu, ...)? Do you have more information on the error?

One other thing that could cause this is not having a settings.json file in the correct position on your file system.
This would be `/home/yourusername/Formula-Student-Driverless-Simulator/settings.json` on linux or `C:\Users\yourusername\Formula-Student-Driverless-Simulator\settings.json` on windows

pyszkaj, 2022-09-27:
Hi @wouter-heerwegh 

I fixed the issue. The problem was actually the folder name which after cloning the repository was called "Formula-Student-Driverless-Simulator-master". Changing it to "Formula-Student-Driverless-Simulator" solved it.

Thank You

closed 2022-09-27 by pyszkaj