FS-Driverless/Formula-Student-Driverless-Simulator #353: Required Process [fsds/ros_bridge-2] has died!

shreyash-99, 2022-12-23:
I am trying to run fsds with ros bridge but i am not able to due to the following error. I even tried reinstalling it but got the same error.
Ubuntu- 18.04 LTS
ROS Melodic

Any help will be much appreciated!
```
roslaunch fsds_ros_bridge fsds_ros_bridge.launch 
... logging to /home/shreyash/.ros/log/a7181916-82f4-11ed-a921-38f9d36a6c99/roslaunch-shreyash-MacBookPro-13697.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://shreyash-MacBookPro:45295/

SUMMARY
========

PARAMETERS
 * /fsds/cameralauncher/host_ip: localhost
 * /fsds/ros_bridge/UDP_control: False
 * /fsds/ros_bridge/competition_mode: False
 * /fsds/ros_bridge/host_ip: localhost
 * /fsds/ros_bridge/manual_mode: False
 * /fsds/ros_bridge/mission_name: trackdrive
 * /fsds/ros_bridge/publish_static_tf_every_n_sec: 1.0
 * /fsds/ros_bridge/track_name: A
 * /fsds/ros_bridge/update_gps_every_n_sec: 0.1
 * /fsds/ros_bridge/update_gss_every_n_sec: 0.01
 * /fsds/ros_bridge/update_imu_every_n_sec: 0.004
 * /fsds/ros_bridge/update_lidar_every_n_sec: 0.1
 * /fsds/ros_bridge/update_odom_every_n_sec: 0.004
 * /rosdistro: melodic
 * /rosversion: 1.14.13

NODES
  /fsds/
    cameralauncher (fsds_ros_bridge/cameralauncher.py)
    ros_bridge (fsds_ros_bridge/fsds_ros_bridge)

auto-starting new master
process[master]: started with pid [13710]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to a7181916-82f4-11ed-a921-38f9d36a6c99
process[rosout-1]: started with pid [13724]
started core service [/rosout]
process[fsds/ros_bridge-2]: started with pid [13727]
process[fsds/cameralauncher-3]: started with pid [13732]
[ INFO] [1671822282.126896837]: Waiting for connection - 
[ INFO] [1671822282.532893066]: Connected to the simulator!
terminate called after throwing an instance of 'std::invalid_argument'
  what():  Failed loading settings.json.rpc::rpc_error during call
localhost
================================================================================REQUIRED process [fsds/ros_bridge-2] has died!
process has died [pid 13727, exit code -6, cmd /home/shreyash/Formula-Student-Driverless-Simulator/ros/devel/lib/fsds_ros_bridge/fsds_ros_bridge __name:=ros_bridge __log:=/home/shreyash/.ros/log/a7181916-82f4-11ed-a921-38f9d36a6c99/fsds-ros_bridge-2.log].
log file: /home/shreyash/.ros/log/a7181916-82f4-11ed-a921-38f9d36a6c99/fsds-ros_bridge-2*.log
Initiating shutdown!
================================================================================
[fsds/cameralauncher-3] killing on exit
[fsds/ros_bridge-2] killing on exit
Traceback (most recent call last):
  File "/usr/bin/rosversion", line 6, in <module>
    from pkg_resources import load_entry_point
  File "/home/shreyash/.local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 3251, in <module>
    @_call_aside
  File "/home/shreyash/.local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 3235, in _call_aside
    f(*args, **kwargs)
  File "/home/shreyash/.local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 3287, in _initialize_master_working_set
    list(map(working_set.add_entry, sys.path))
  File "/home/shreyash/.local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 623, in add_entry
    for dist in find_distributions(entry, True):
  File "/home/shreyash/.local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2065, in find_on_path
    for dist in factory(fullpath):
  File "/home/shreyash/.local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2135, in distributions_from_metadata
    root, entry, metadata, precedence=DEVELOP_DIST,
  File "/home/shreyash/.local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2592, in from_location
    py_version=py_version, platform=platform, **kw
  File "/home/shreyash/.local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2994, in _reload_version
    md_version = self._get_version()
  File "/home/shreyash/.local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2772, in _get_version
    version = _version_from_file(lines)
  File "/home/shreyash/.local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2556, in _version_from_file
    line = next(iter(version_lines), '')
  File "/home/shreyash/.local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2767, in _get_metadata
    for line in self.get_metadata_lines(name):
  File "/home/shreyash/.local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 1911, in get_metadata_lines
    return yield_lines(self.get_metadata(name))
  File "/home/shreyash/.local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 1898, in get_metadata
    metadata = f.read()
  File "/usr/lib/python2.7/codecs.py", line 311, in decode
    def decode(self, input, final=False):
KeyboardInterrupt
Traceback (most recent call last):
  File "/home/shreyash/Formula-Student-Driverless-Simulator/ros/src/fsds_ros_bridge/scripts/cameralauncher.py", line 38, in <module>
    launch.start()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/scriptapi.py", line 107, in start
    self.parent.start(auto_terminate=False)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/parent.py", line 305, in start
    self._start_infrastructure()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/parent.py", line 254, in _start_infrastructure
    self._load_config()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/parent.py", line 157, in _load_config
    roslaunch_strs=self.roslaunch_strs, verbose=self.verbose)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/config.py", line 451, in load_config_default
    load_roscore(loader, config, verbose=verbose)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/config.py", line 94, in load_roscore
    loader.load(f_roscore, config, core=True, verbose=verbose)    
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 761, in load
    self._load_launch(launch, ros_config, is_core=core, filename=filename, argv=argv, verbose=verbose)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 733, in _load_launch
    self._recurse_load(ros_config, launch.childNodes, self.root_context, None, is_core, verbose)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 669, in _recurse_load
    default_machine, is_core, verbose)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 679, in _recurse_load
    self._param_tag(tag, context, ros_config, verbose=verbose)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 96, in call
    return f(*args, **kwds)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 286, in _param_tag
    "Invalid <param> tag: %s. \n\nParam xml is %s"%(e, tag.toxml()))
roslaunch.xmlloader.XmlParseException: Invalid <param> tag: Cannot load command parameter [rosversion]: command [['rosversion', 'roslaunch']] returned with code [1]. 

Param xml is <param command="rosversion roslaunch" name="rosversion"/>
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete

```

wouter-heerwegh, 2022-12-25:
Hi @shreyash-99

I don't think we support Mac OS out of the box, but seeing from the error logs, it seems to fail on trying to use `localhost`. Could you try to use the `host` argument to 127.0.0.1 or the assigned ip when launching the fsds_ros_bridge?

MohakVyas, 2023-01-02:
Hey @wouter-heerwegh 
Tried this suggestion by changing the argument of host from localhost to 127.0.0.1 in the fsds_ros_bridge.cpp and the fsds_ros_bridge.launch files. But still receiving the same error when running fsds ros bridge.
running ros noetic on ubuntu 20.04

wouter-heerwegh, 2023-01-02:
@MohakVyas, are you using the v2.1.0 executable with the latest code? If so, can you try to run `git checkout v2.1.0` and rebuild and install Airsim?

ThoroMech, 2023-01-02:
Hi @wouter-heerwegh and @MohakVyas 

I am new to the simulator (Ubuntu 22.04 with ros2 humble) and am getting the same error:

fsds_ros2_bridge-3] terminate called after throwing an instance of 'std::invalid_argument'
[fsds_ros2_bridge-3]   what():  Failed loading settings.json.rpc::rpc_error during call

My debugging so far (Apologies if this is obvious/not helpful):

my error is coming from line 18 of airsim_ros_wrapper.cpp:

18      std::string settings_text = airsim_client_.getSettingsString();

getSettingsString() is declared in lines 379-382 of RpcLibClientBase.cpp:

379 std::string RpcLibClientBase::getSettingsString() const
380 {
381     return pimpl_->client.call("getSettingsString").as<std::string>();
382 }

wouter-heerwegh, 2023-01-02:
Hi @ThoroMech,

This is a mismatch in versions of the rpc server. This issue is introduced by #324, where the functionality of the rpc api was updated. V2.1.0 Does not contain the functionality to get the settings in that way. #333 Should fix this.

For now, it is best to use the source code at tag v2.1.0 or use any commit between v2.1.0 and 3047f0e24dce71b127577eee60150db4540fcfde

To switch to the fsds_ros_bridge/fsds_ros2_bridge code of v2.1.0 use `git checkout tags/v2.1.0`. Your orther options for now is to run it in unreal engine (which I don't advice if you don't need to use it).

I'll try to release v2.2.0 this weekend

ThoroMech, 2023-01-03:
Thankyou @wouter-heerwegh 

I am new to a lot of this, your effort is appreciated.

My understanding now is: prior to #327 AirSim/setup.sh will not work on 22.04 due to clang compatibility, and as you stated ros2_bridge has the rpc error after #324.  

I have tried a couple different things but am unable to build the ros2_bridge from <#324 with airsim setup >#327. Is there a way to overcome this without reverting to a compatible ubuntu?

I might be best waiting for v2.20.

wouter-heerwegh, 2023-01-03:
Hi @ThoroMech,

You could apply the changes from #327 manually to your local prpject to get it up and running. I'm still having an issue in the current build (#289), if I can't get it fixed by the weekend, I'll release v2.2.0.

wouter-heerwegh, 2023-01-15:
Sorry that it took a bit longer, but release [v2.2.0](https://github.com/FS-Driverless/Formula-Student-Driverless-Simulator/releases/tag/v2.2.0) is available. Test it out and feel free to give feedback.

If the above issue is fixed you can close this issue.

ThoroMech, 2023-01-21:
Thanks Again @wouter-heerwegh,

Having some PC issues at the moment, hopefully have them sorted and be able to test ROS2 bridge by next weekend.

ThoroMech, 2023-02-05:
Thanks Wouter

Everything is working well for me so far. The update has also fixed the binary loading to black screen.
only thing i have found is FSDS.sh needs to be made executable before using since the update.

wouter-heerwegh, 2023-02-06:
@ThoroMech thanks for the feedback. I uploaded a new zip file for the linux version with the FSDS.sh script as an executable.

I'm closing this issue as I expect it is fixed, if not, feel free to reopen.

closed 2023-02-06 by wouter-heerwegh