FS-Driverless/Formula-Student-Driverless-Simulator #354: Unable to change camera Image Type

Lamfurst, 2023-02-22:
Hey guys! 
I Install the fsds from release binaries, and I change the settings.json as following.
[image]

Seems that the position change works but ImageType does not. I wrote a simple subscriber to catch the information in /fsds/cam2 topic using the following code as following:
[image]
But the ROS_INFO gives these information which mismatches with the settings.json (both height, width and encoding type).
[image]

I am new to ROS and I'd much appreciate it if you could tell me how should I deal with this problem. 
Also, please tell me if there is a better way to write a subscriber function that catches the msg only once. Thx!

masskro0, 2023-03-21:
Did you restart the simulator? I usually restart it after every change in the settings.json file (not sure if this is necessary, but it works).

Lamfurst, 2023-08-09:
Thanks for your response. I realized that this is because I did not change the settings.json on both sides

closed 2023-08-09 by Lamfurst