FS-Driverless/Formula-Student-Driverless-Simulator #355: REQUIRED process [fsds/ros_bridge-2] has died!

AceyGit, 2023-03-27:
Hey everyone, I'm a very fresh face in the AV/DV world and have started my final year project on it. I am getting a very similar issue to #353 and can't seem to work it out, as I have restarted a million times, reverted back to v2.1, attempted (and failed) compiling the simulator and so much more but no luck. 

I am running:
Ubuntu - 18.04.5 LTS on WSL
ROS Melodic
Pre-compiled fsds executable file at C:/users/username/Formula-Student-Driverless-Simulator

Having followed the setup guide to the best of my knowledge, I have been getting the following error when I try and launch the ROS bridge and didn't particularly understand the solution from the other issue closed.

```
roslaunch fsds_ros_bridge fsds_ros_bridge.launch
... logging to /home/danielwsl/.ros/log/bd17725a-cccf-11ed-b745-00155d3b13c4/roslaunch-Daniel-8689.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://Daniel:46769/

SUMMARY
========

PARAMETERS
 * /fsds/cameralauncher/host_ip: localhost
 * /fsds/ros_bridge/UDP_control: False
 * /fsds/ros_bridge/competition_mode: False
 * /fsds/ros_bridge/host_ip: localhost
 * /fsds/ros_bridge/manual_mode: False
 * /fsds/ros_bridge/mission_name: trackdrive
 * /fsds/ros_bridge/publish_static_tf_every_n_sec: 1.0
 * /fsds/ros_bridge/track_name: A
 * /fsds/ros_bridge/update_gps_every_n_sec: 0.1
 * /fsds/ros_bridge/update_gss_every_n_sec: 0.01
 * /fsds/ros_bridge/update_imu_every_n_sec: 0.004
 * /fsds/ros_bridge/update_lidar_every_n_sec: 0.1
 * /fsds/ros_bridge/update_odom_every_n_sec: 0.004
 * /rosdistro: melodic
 * /rosversion: 1.14.13

NODES
  /fsds/
    cameralauncher (fsds_ros_bridge/cameralauncher.py)
    ros_bridge (fsds_ros_bridge/fsds_ros_bridge)

auto-starting new master
process[master]: started with pid [8699]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to bd17725a-cccf-11ed-b745-00155d3b13c4
process[rosout-1]: started with pid [8710]
started core service [/rosout]
process[fsds/ros_bridge-2]: started with pid [8716]
process[fsds/cameralauncher-3]: started with pid [8718]
[ INFO] [1679942812.600812779]: Waiting for connection -
localhost
started roslaunch server http://Daniel:41801/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.13

NODES

ROS_MASTER_URI=http://localhost:11311
process[fsds/camera/cam2-1]: started with pid [8747]
process[fsds/camera/cam1-2]: started with pid [8748]
Waiting for connection -
Waiting for connection -
terminate called after throwing an instance of 'std::runtime_error'
  what():  Failed connecting to RPC server (airsim). Is the simulator running?
================================================================================REQUIRED process [fsds/ros_bridge-2] has died!
process has died [pid 8716, exit code -6, cmd /home/danielwsl/Formula-Student-Driverless-Simulator/ros/devel/lib/fsds_ros_bridge/fsds_ros_bridge __name:=ros_bridge __log:=/home/danielwsl/.ros/log/bd17725a-cccf-11ed-b745-00155d3b13c4/fsds-ros_bridge-2.log].
log file: /home/danielwsl/.ros/log/bd17725a-cccf-11ed-b745-00155d3b13c4/fsds-ros_bridge-2*.log
Initiating shutdown!
================================================================================
[fsds/cameralauncher-3] killing on exit
[fsds/ros_bridge-2] killing on exit
[fsds/camera/cam1-2] killing on exit
[fsds/camera/cam2-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

Any help would be greatly appreciated,

Thanks!

wouter-heerwegh, 2023-03-28:
Hi @AceyGit, I'm guessing you're running ROS in WSL? WSL uses a network bridge, so the ROS bridge can have troubles finding the simulator. Try to get your WSL host ip, for instance with this command:
```
cat /etc/resolv.conf | grep nameserver | cut -d ' ' -f 2
```
And try to use the `host` parameter of the ROS bridge (this value defaults to `localhost`)

AceyGit, 2023-03-28:
Hi @wouter-heerwegh,

Thank you for the help, it has gotten me over the first hurdle and mainly solved my issues. I can now connect successfully but when I do connect with Rviz in which it appears that camera data is not being published, inclusive of when I run 
`rostopic /fsds/camera/cam1`

[image]

I am not moving the vehicle as of yet, and I want to make sure that would not affect this. As well, I would like your advice as to where to start/ how to move the vehicle for a person new to this 'software', seeing as a keyboard does not work whilst on the bridge.

Any help or pointers in the right direction is greatly appreciated. Thanks!

wouter-heerwegh, 2023-03-28:
I do see `/fsds/cam1` and `/fsds/cam2`, do these contain the images? If so I'll update the documentation to contain the correct topic names

AceyGit, 2023-03-28:
Yep, that does appear to be the case!

wouter-heerwegh, 2023-03-28:
The code in the bridges has been updated to properly follow the information in the docs (#356)

PR #356 (2023-03-28): Change camera prefix to follow documentation and to follow lidar naming convention
Fixes #355
closed 2023-03-28 by wouter-heerwegh