FS-Driverless/Formula-Student-Driverless-Simulator #363: Run the simulation in Ubuntu 22.04.2 ARM64

santysangro, 2023-11-14:
Hi!
So I am currently trying to download the simulator in Ubuntu 22.04.2 ARM64, and I cannot seem to make it work. Can someone give me a little step by step on how to make it work? It would be greatly appreciated.
Thanks!

wouter-heerwegh, 2023-11-14:
Hi @santysangro,

I'm assuming you're trying to run the binaries? Those are built for x86, we currently do not build them for ARM. If you would want to run it on ARM, you would have to recompile it from source. The documentation should explain how to do so, but if you have any problems with it, feel free to hit me up.

santysangro, 2023-11-14:
Hey @wouter-heerwegh 
Yes, I am trying to run the binaries. Do you mean by recompile it from source using the Unreal Engine editor? I have also tried, but it also does not seem to work.

wouter-heerwegh, 2023-11-14:
Yes, you would need to use Unreal Engine, and package it for your system. Did you manage to build Airsim and does it open the project in Unreal Engine?

santysangro, 2023-11-14:
I did not. I cloned the Unreal Engine repo but when trying to run the ./Setup.sh command, it gives me Exec format error. How could I fix this?

wouter-heerwegh, 2023-11-14:
Would be hard to say without any error trace. Would you be able to put the terminal output here?

santysangro, 2023-11-14:
sure!
Registering git hooks... (this will override existing ones!)
Fixing inconsistent case in filenames.
Setting up Mono
Engine/Build/BatchFiles/Linux/GitDependencies.sh: line 25: /home/parallels/UnrealEngine/Engine/Build/BatchFiles/Linux/../../../Binaries/ThirdParty/Mono/Linux/bin/mono: cannot execute binary file: Exec format error
Result: 126

wouter-heerwegh, 2023-11-14:
I think you would have to install mono first see [this](https://linuxhint.com/install-use-mono-ubuntu-22-04/). Could you check if it works after that?

santysangro, 2023-11-14:
I have this mono version, but I get the same error.
Mono JIT compiler version 6.12.0.200 (tarball Tue Jul 11 21:45:16 UTC 2023)
Copyright (C) 2002-2014 Novell, Inc, Xamarin Inc and Contributors. www.mono-project.com
	TLS:           __thread
	SIGSEGV:       normal
	Notifications: epoll
	Architecture:  arm64
	Disabled:      none
	Misc:          softdebug 
	Interpreter:   yes
	LLVM:          yes(610)
	Suspend:       preemptive
	GC:            sgen (concurrent by default)

santysangro, 2023-11-14:
I also get this warning when running sudo apt update 
W: https://download.mono-project.com/repo/ubuntu/dists/stable-focal/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
maybe it helps.

wouter-heerwegh, 2023-11-14:
I think it is trying to use a mono executable in this path `/home/parallels/UnrealEngine/Engine/Build/BatchFiles/Linux/../../../Binaries/ThirdParty/Mono/Linux/bin/mono`. Could you check that the file is executable? I.e. `ls -la /home/parallels/UnrealEngine/Engine/Build/BatchFiles/Linux/../../../Binaries/ThirdParty/Mono/Linux/bin/mono` and seeing if it contains x's. The deprecation has to do with the apt repository for mono, but should not be an issue

santysangro, 2023-11-14:
Yeah, I get this:
-rwxrwxr-x 1 parallels parallels 19432917 Nov 14 22:48 /home/parallels/UnrealEngine/Engine/Build/BatchFiles/Linux/../../../Binaries/ThirdParty/Mono/Linux/bin/mono

wouter-heerwegh, 2023-11-14:
Is this file executable? `Engine/Binaries/DotNET/GitDependencies.exe`

santysangro, 2023-11-14:
Yes, the file is running,

wouter-heerwegh, 2023-11-14:
Could you try to run `mono Engine/Binaries/DotNET/GitDependencies.exe` manually?

santysangro, 2023-11-14:
Yes, it also works.

wouter-heerwegh, 2023-11-14:
So the `Setup.sh` script runs fine now?

santysangro, 2023-11-15:
No, I still get the same eror:
Registering git hooks... (this will override existing ones!)
Fixing inconsistent case in filenames.
Setting up Mono
Engine/Build/BatchFiles/Linux/GitDependencies.sh: line 25: /home/parallels/UnrealEngine/Engine/Build/BatchFiles/Linux/../../../Binaries/ThirdParty/Mono/Linux/bin/mono: cannot execute binary file: Exec format error
Result: 126

wouter-heerwegh, 2023-11-16:
I don't immediately have an idea then. It could be good to open a post [here](https://forums.unrealengine.com/categories?tag=unreal-engine) since it seems to be an unreal engine issue. I've seen some posts like [this one](https://forums.unrealengine.com/t/packaging-linux-arm64-exec-format-error/385806) that experience a similar issue, but not sure if that will fix it in your case.