FS-Driverless/Formula-Student-Driverless-Simulator #364: Issue when restarting the bridge & FSDS.exe (WSL)

Juphex, 2023-12-20:
Hello,
when I start FSDS.exe and connect it to the bridge in WSL everything works fine.

However, when restarting the exe and the launch file of the bridge I get this error below.
It gets solved when I fully restart my computer :/
Is this a known bug or is there a fix?

```
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.178.53:60761/

SUMMARY
========

PARAMETERS
 * /fsds/cameralauncher/host_ip: localhost
 * /fsds/ros_bridge/UDP_control: False
 * /fsds/ros_bridge/competition_mode: False
 * /fsds/ros_bridge/host_ip: localhost
 * /fsds/ros_bridge/manual_mode: False
 * /fsds/ros_bridge/mission_name: trackdrive
 * /fsds/ros_bridge/publish_static_tf_every_n_sec: 1.0
 * /fsds/ros_bridge/track_name: A
 * /fsds/ros_bridge/update_gps_every_n_sec: 0.1
 * /fsds/ros_bridge/update_gss_every_n_sec: 0.01
 * /fsds/ros_bridge/update_imu_every_n_sec: 0.004
 * /fsds/ros_bridge/update_lidar_every_n_sec: 0.1
 * /fsds/ros_bridge/update_odom_every_n_sec: 0.004
 * /rosdistro: melodic
 * /rosversion: 1.14.13

NODES
  /fsds/
    cameralauncher (fsds_ros_bridge/cameralauncher.py)
    ros_bridge (fsds_ros_bridge/fsds_ros_bridge)

auto-starting new master
process[master]: started with pid [10794]
ROS_MASTER_URI=http://192.168.178.53:11311

setting /run_id to 8eafb92a-9f76-11ee-9763-e84e06a92f9a
process[rosout-1]: started with pid [10805]
started core service [/rosout]
process[fsds/ros_bridge-2]: started with pid [10812]
process[fsds/cameralauncher-3]: started with pid [10826]
Ground Speed Sensor
Gps
Imu
Lidar
Lidar
Waiting for connection -
started roslaunch server http://192.168.178.53:60839/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.13

NODES

ROS_MASTER_URI=http://192.168.178.53:11311
process[fsds/camera/cam2-1]: started with pid [10851]
process[fsds/camera/cam1-2]: started with pid [10854]
Waiting for connection -
Waiting for connection -
[cambridge cam2] Exception raised by the API, something went wrong.
rpc::timeout: Timeout of 5000ms while calling RPC function 'getServerVersion'
[cambridge cam1] Exception raised by the API, something went wrong.
rpc::timeout: Timeout of 5000ms while calling RPC function 'getServerVersion'
================================================================================REQUIRED process [fsds/camera/cam2-1] has died!
process has died [pid 10851, exit code 1, cmd /home/chris/Formula-Student-Driverless-Simulator/ros/devel/lib/fsds_ros_bridge/fsds_ros_bridge_camera _depthcamera:=False _camera_name:=cam2 _framerate:=15 _host_ip:=localhost __name:=cam2 __log:=/home/chris/.ros/log/8eafb92a-9f76-11ee-9763-e84e06a92f9a/fsds-camera-cam2-1.log].
log file: /home/chris/.ros/log/8eafb92a-9f76-11ee-9763-e84e06a92f9a/fsds-camera-cam2-1*.log
Initiating shutdown!
================================================================================
[fsds/camera/cam1-2] killing on exit
[fsds/camera/cam2-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
================================================================================REQUIRED process [fsds/cameralauncher-3] has died!
process has finished cleanly
log file: /home/chris/.ros/log/8eafb92a-9f76-11ee-9763-e84e06a92f9a/fsds-cameralauncher-3*.log
Initiating shutdown!
================================================================================
[fsds/cameralauncher-3] killing on exit
[fsds/ros_bridge-2] killing on exit

```

wouter-heerwegh, 2023-12-20:
Hi @Juphex,

It looks a bit as if the RPC server port is still opened. This is used to transfer data from the sim to the ros bridge.

I think the port is 41451, could you check if a process is still keeping this port opened? And if so, could you kill the process and restart both the sim and the ros bridge?

Juphex, 2023-12-20:
There are no processes running, I assume. Therefore, I cannot kill any process.
[image]

Btw: Is there a "correct" way to close the simulation?
Even restarting WSL (wsl --shutdown) isn't enough. I have to restart my whole computer. :/

wouter-heerwegh, 2023-12-21:
Is this also the case on the Windows side?

I've never really had this issue and mostly used wsl to connect to the sim. Could you check if shutting down the bridge first, and then the unreal executable helps?

Juphex, 2023-12-21:
Apprently, shutting down the bridge first, and then the unreal executable works.

wouter-heerwegh, 2023-12-21:
Okay, good to know. Could you verify that if you shut it down incorrectly that there is no port opened on 41451 in Windows (not wsl)?

Juphex, 2024-01-04:
Not really, I am sorry.

Juphex, 2024-01-04:
Yes, I could verify that the port was still used by a process (presumably the simulation). Killing the program on that port helped.