FS-Driverless/Formula-Student-Driverless-Simulator #366: Cannot clone the repository (Git LFS is over its data quota)

vrepetskyi, 2024-03-11:
```
git clone https://github.com/FS-Driverless/Formula-Student-Driverless-Simulator.git
Cloning into 'Formula-Student-Driverless-Simulator'...
remote: Enumerating objects: 8876, done.
remote: Counting objects: 100% (156/156), done.
remote: Compressing objects: 100% (48/48), done.
remote: Total 8876 (delta 63), reused 142 (delta 58), pack-reused 8720
Receiving objects: 100% (8876/8876), 688.11 MiB | 12.31 MiB/s, done.
Resolving deltas: 100% (4634/4634), done.
Downloading UE4Project/Content/Blocks/1M_Cube.uasset (85 KB)
Error downloading object: UE4Project/Content/Blocks/1M_Cube.uasset (3a702be): Smudge error: Error downloading UE4Project/Content/Blocks/1M_Cube.uasset (3a702be2ee121e56cb97f116be968e633774cd91d12207e43f7bc09f6383f428): batch response: This repository is over its data quota. Account responsible for LFS bandwidth should purchase more data packs to restore access.

Errors logged to /home/vrepetskyi/classes/6/rob/Formula-Student-Driverless-Simulator/.git/lfs/logs/20240311T111249.002330546.log
Use `git lfs logs last` to view the log.
error: external filter 'git-lfs filter-process' failed
fatal: UE4Project/Content/Blocks/1M_Cube.uasset: smudge filter lfs failed
warning: Clone succeeded, but checkout failed.
You can inspect what was checked out with 'git status'
and retry with 'git restore --source=HEAD :/'
```

wouter-heerwegh, 2024-03-11:
Hi @vrepetskyi,

I just looked into it, it seems like we got 667 clones in just 1 day, even though there were 3 unique cloners on the 1st of March.
[image]

This is much more than on a regular day: 
[image]

I'm not sure we can really do anything about that at the moment, but billing shows that the quota should reset today/tomorrow.

If anyone is running some testing scripts, please consider cloning the repo to a local directory and using your local copy to run tests.

RafaGuil, 2024-04-29:
Hi, I have the same issue: 

` C:\Users\Rafa Guil Valero>git clone https://github.com/FS-Driverless/Formula-Student-Driverless-Simulator.git
Cloning into 'Formula-Student-Driverless-Simulator'...
remote: Enumerating objects: 8876, done.
remote: Counting objects: 100% (156/156), done.
remote: Compressing objects: 100% (48/48), done.
remote: Total 8876 (delta 63), reused 142 (delta 58), pack-reused 8720Receiving objects: 100% (8876/8876), 682.89 MiB | Receiving objects: 100% (8876/8876), 688.11 MiB | 11.16 MiB/s, done.

Resolving deltas: 100% (4611/4611), done.
Updating files: 100% (739/739), done.
Downloading UE4Project/Content/Blocks/1M_Cube.uasset (85 KB)
Error downloading object: UE4Project/Content/Blocks/1M_Cube.uasset (3a702be): Smudge error: Error downloading UE4Project/Content/Blocks/1M_Cube.uasset (3a702be2ee121e56cb97f116be968e633774cd91d12207e43f7bc09f6383f428): batch response: This repository is over its data quota. Account responsible for LFS bandwidth should purchase more data packs to restore access.

Errors logged to 'C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\.git\lfs\logs\20240429T115759.0683178.log'.
Use `git lfs logs last` to view the log.
error: external filter 'git-lfs filter-process' failed
fatal: UE4Project/Content/Blocks/1M_Cube.uasset: smudge filter lfs failed
warning: Clone succeeded, but checkout failed.
You can inspect what was checked out with 'git status'
and retry with 'git restore --source=HEAD :/'

C:\Users\Rafa Guil Valero>cd Formula-Student-Driverless-Simulator

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator>git lfs pull
batch response: This repository is over its data quota. Account responsible for LFS bandwidth should purchase more data packs to restore access.
batch response: This repository is over its data quota. Account responsible for LFS bandwidth should purchase more data packs to restore access.
batch response: This repository is over its data quota. Account responsible for LFS bandwidth should purchase more data packs to restore access.
batch response: This repository is over its data quota. Account responsible for LFS bandwidth should purchase more data packs to restore access.
Failed to fetch some objects from 'https://github.com/FS-Driverless/Formula-Student-Driverless-Simulator.git/info/lfs' `

Could you reset the data quota of the repository? Thanks in advance

RafaGuil, 2024-04-30:
@wouter-heerwegh, I really need lfs files, I have to edit car's physics. What can I do to get those files? Is there any other way to get them?

wouter-heerwegh, 2024-04-30:
@RafaGuil, as long as the data is not paid, we can't clone the LFS files anymore. I still have them locally, so as a temporary workaround, I'll upload them [here]().

@SijmenHuizenga, I see that all billing is forwarded towards you. Are you directly paying the 5 euro a month or is this still payed by FSOnline? I saw that there was also no payment history for February and March.

If payments stop, I guess we'll have to move to a different solution than LFS.

SijmenHuizenga, 2024-04-30:
@RafaGuil can you try again? Bandwidth should be reset now.

Github stopped sending invoices because:
[image]
Re-submitting organization details fixed it, invoicing should continue.

Yes, I pay the datapacks. I would love to move to a free place to host git+lfs, but last time i checked (year ago) I couldn't find any. Suggestions are more than welcome.

RafaGuil, 2024-04-30:
Yeah, now I have the lfs files
[image]

But I still have the same problem. When I open FSOnline.uproject I have this error:
[image]

wouter-heerwegh, 2024-04-30:
@SijmenHuizenga, indeed, I also didn't find any nice alternate solutions. We could use a self hosted server, but it would be less friendly to use for teams.

@RafaGuil don't forget to compile AirSim. There are currently some issues when building the UE project on Ubuntu 20.04 and up, because of changing glibc versions and the toolchain of UE4.27. I would highly recommend working on the UE code in windows.

RafaGuil, 2024-04-30:
@wouter-heerwegh Also I have errors in Windows...
**********************************************************************
** Visual Studio 2019 Developer Command Prompt v16.11.35
** Copyright (c) 2021 Microsoft Corporation
**********************************************************************

C:\Program Files (x86)\Microsoft Visual Studio\2019\Community>cd ../../../..

C:\>cd Users

C:\Users>cd "Rafa Guil Valero"

C:\Users\Rafa Guil Valero>cd Formula-Student-Driverless-Simulator

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator>cd AirSim

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>build.cmd

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>REM //---------- set up variable ----------

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>setlocal

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>set ROOT_DIR=C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim\

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>REM // Check command line arguments

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>set "noFullPolyCar="

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>set "buildMode="

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>REM //check VS version

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>if "16.0" == "" (
echo(
 echo oh oh... You need to run this command from x64 Native Tools Command Prompt for VS 2019.
 goto :buildfailed_nomsg
)

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>if "16.0" LSS "16.0" (
echo(
 echo Hello there! We just upgraded AirSim to Unreal Engine 4.24 and Visual Studio 2019.
 echo Here are few easy steps for upgrade so everything is new and shiny:
 echo https://github.com/Microsoft/AirSim/blob/master/docs/unreal_upgrade.md
 goto :buildfailed_nomsg
)

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>if "" == "" goto noargs

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>chdir /d C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim\

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>REM //---------- Check cmake version ----------

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>CALL check_cmake.bat

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>REM //---------- set up variable ----------

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>setlocal

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>set ROOT_DIR=C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim\

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>set cmake_minversion_minmaj="   3.   9"

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>set "cmake_version=    .    "

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>where /q cmake

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>if 0 EQU 0 (
for /F "tokens=3" %a in ('cmake --version | find "cmake version"') do set cmake_version=%a
 if "    .    " == "" (
echo Unable to get version of cmake.  1>&2
[... 109 lines cut ...]
                               = Generates Eclipse CDT 4.0 project files.
  Eclipse CDT4 - Ninja         = Generates Eclipse CDT 4.0 project files.
  Eclipse CDT4 - Unix Makefiles= Generates Eclipse CDT 4.0 project files.

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim\external\rpclib\build>if "" == "--Debug" (cmake --build . --config Debug )  else if "" == "--Release" (cmake --build . --config Release )  else (
cmake --build .
 cmake --build . --config Release
)
Error: could not load cache
Error: could not load cache

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim\external\rpclib\build>if ERRORLEVEL 1 goto :buildfailed

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim\external\rpclib\build>echo(

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim\external\rpclib\build>echo #### Build failed - see messages above.  1>&2
#### Build failed - see messages above.

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim\external\rpclib\build>chdir /d C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim\

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>exit /b 1

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>

I have Windows 11, UE 4.27 and VS 2019

wouter-heerwegh, 2024-04-30:
@RafaGuil I'm moving your last comment to a new issue, as this is not related to the LFS issues.