FS-Driverless/Formula-Student-Driverless-Simulator #367: ROS2: Cannot read /testing_only/track through fsds_ros2_bridge

froxec, 2024-04-09:
OS: ROS2 Humble
Problem description: 
Whenever I try to read /testing_only/track message it shows nothing.
Message flow diagram:
 FSDS -> fsds_ros2_bridge -> ros2 topic echo command/c++ subscriber.

Prior investigation conclusions:
1. I think that fsds_ros2_bridge is at fault. 
2. /testing_only/track is a **latched** message.
3. Latched messages in ROS2 work by setting proper QoS profile: "The “durability” policy “transient local”, combined with any depth, provides functionality similar to that of “latching” publishers." [Source](https://docs.ros.org/en/humble/Concepts/Intermediate/About-Quality-of-Service-Settings.html#comparison-to-ros-1)
4. Requirement from point 3. is not met because message qos-durability is set to volatile (which means that samples don't persist on the publisher side) [SOURCE](https://docs.ros.org/en/humble/Concepts/Intermediate/About-Quality-of-Service-Settings.html#qos-policies).

Point 4. evidence (output of `ros2 topic info --verbose /testing_only/track`):
> Type: fs_msgs/msg/Track 
> Publisher count: 1
> Node name: ros_bridge
> Node namespace: /
> Topic type: fs_msgs/msg/Track
> Endpoint type: PUBLISHER
> QoS profile:
>   Reliability: RELIABLE
>   History (Depth): UNKNOWN
>   Durability: VOLATILE
>   Lifespan: Infinite
>   Deadline: Infinite
>   Liveliness: AUTOMATIC
>   Liveliness lease duration: Infinite

Possible solution: **Change fsds_ros2_bridge QoS profile for /testing_only/track message (as described in point 3.)**
Is that possible to change QoS profile in fsds_ros2_bridge?

wouter-heerwegh, 2024-04-09:
Hi @froxec,

I currently do not have a ROS2 environment set up. If you'd like to have this implemented quickly and would like to contribute, feel free to open a PR. Otherwise I will have to find some time.

froxec, 2024-04-15:
Hello @wouter-heerwegh,

I hotfixed this by writing the whole track (cones colors + locations) to json file (through Python client) and then simply loading it in c++ node.

PR #376 (2025-02-07): ROS2 Namespace and testing_only/track latching
I'm finally porting my personal code to ROS2, I saw that `/fsds/` was never prepended to any topic (while it is in ROS1).

I added the fsds namespace to the bridge node to align with the documentation and ROS1 and also made `testing_only/track` transient local. If your subscriber is also transient local, it should act like a latching topic now. This fixes #367.
closed 2025-02-07 by wouter-heerwegh