FS-Driverless/Formula-Student-Driverless-Simulator #368: Windows 11 AirSim not building

wouter-heerwegh, 2024-04-30:
@wouter-heerwegh Also I have errors in Windows...
**********************************************************************
** Visual Studio 2019 Developer Command Prompt v16.11.35
** Copyright (c) 2021 Microsoft Corporation
**********************************************************************

C:\Program Files (x86)\Microsoft Visual Studio\2019\Community>cd ../../../..

C:\>cd Users

C:\Users>cd "Rafa Guil Valero"

C:\Users\Rafa Guil Valero>cd Formula-Student-Driverless-Simulator

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator>cd AirSim

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>build.cmd

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>REM //---------- set up variable ----------

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>setlocal

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>set ROOT_DIR=C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim\

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>REM // Check command line arguments

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>set "noFullPolyCar="

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>set "buildMode="

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>REM //check VS version

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>if "16.0" == "" (
echo(
 echo oh oh... You need to run this command from x64 Native Tools Command Prompt for VS 2019.
 goto :buildfailed_nomsg
)

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>if "16.0" LSS "16.0" (
echo(
 echo Hello there! We just upgraded AirSim to Unreal Engine 4.24 and Visual Studio 2019.
 echo Here are few easy steps for upgrade so everything is new and shiny:
 echo https://github.com/Microsoft/AirSim/blob/master/docs/unreal_upgrade.md
 goto :buildfailed_nomsg
)

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>if "" == "" goto noargs

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>chdir /d C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim\

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>REM //---------- Check cmake version ----------

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>CALL check_cmake.bat

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>REM //---------- set up variable ----------

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>setlocal

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>set ROOT_DIR=C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim\

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>set cmake_minversion_minmaj="   3.   9"

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>set "cmake_version=    .    "

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>where /q cmake

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>if 0 EQU 0 (
for /F "tokens=3" %a in ('cmake --version | find "cmake version"') do set cmake_version=%a
 if "    .    " == "" (
echo Unable to get version of cmake.  1>&2
[... 111 lines cut ...]
  Eclipse CDT4 - Unix Makefiles= Generates Eclipse CDT 4.0 project files.

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim\external\rpclib\build>if "" == "--Debug" (cmake --build . --config Debug )  else if "" == "--Release" (cmake --build . --config Release )  else (
cmake --build .
 cmake --build . --config Release
)
Error: could not load cache
Error: could not load cache

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim\external\rpclib\build>if ERRORLEVEL 1 goto :buildfailed

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim\external\rpclib\build>echo(

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim\external\rpclib\build>echo #### Build failed - see messages above.  1>&2
#### Build failed - see messages above.

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim\external\rpclib\build>chdir /d C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim\

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>exit /b 1

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>

I have Windows 11, UE 4.27 and VS 2019

_Originally posted by @RafaGuil in https://github.com/FS-Driverless/Formula-Student-Driverless-Simulator/issues/366#issuecomment-2085045502_

wouter-heerwegh, 2024-04-30:
@RafaGuil, I've never built on Windows 11 before, I'll try to check it out

RafaGuil, 2024-04-30:
@wouter-heerwegh, Same error :(
**********************************************************************
** Visual Studio 2019 Developer Command Prompt v16.11.35
** Copyright (c) 2021 Microsoft Corporation
**********************************************************************
[vcvarsall.bat] Environment initialized for: 'x64'

C:\Program Files (x86)\Microsoft Visual Studio\2019\Community>cd ../../../..

C:\>cd Users

C:\Users>cd "Rafa Guil Valero"

C:\Users\Rafa Guil Valero>cd Formula-Student-Driverless-Simulator

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator>git submodule update --init --recursive

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator>cd AirSim

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>build.cmd

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>REM //---------- set up variable ----------

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>setlocal

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>set ROOT_DIR=C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim\

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>REM // Check command line arguments

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>set "noFullPolyCar="

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>set "buildMode="

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>REM //check VS version

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>if "16.0" == "" (
echo(
 echo oh oh... You need to run this command from x64 Native Tools Command Prompt for VS 2019.
 goto :buildfailed_nomsg
)

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>if "16.0" LSS "16.0" (
echo(
 echo Hello there! We just upgraded AirSim to Unreal Engine 4.24 and Visual Studio 2019.
 echo Here are few easy steps for upgrade so everything is new and shiny:
 echo https://github.com/Microsoft/AirSim/blob/master/docs/unreal_upgrade.md
 goto :buildfailed_nomsg
)

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>if "" == "" goto noargs

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>chdir /d C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim\

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>REM //---------- Check cmake version ----------

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>CALL check_cmake.bat

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>REM //---------- set up variable ----------

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>setlocal

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>set ROOT_DIR=C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim\

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>set cmake_minversion_minmaj="   3.   9"

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>set "cmake_version=    .    "

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>where /q cmake

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>if 0 EQU 0 (
[... 110 lines cut ...]
                               = Generates Eclipse CDT 4.0 project files.
  Eclipse CDT4 - MinGW Makefiles
                               = Generates Eclipse CDT 4.0 project files.
  Eclipse CDT4 - Ninja         = Generates Eclipse CDT 4.0 project files.
  Eclipse CDT4 - Unix Makefiles= Generates Eclipse CDT 4.0 project files.

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim\external\rpclib\build>if "" == "--Debug" (cmake --build . --config Debug )  else if "" == "--Release" (cmake --build . --config Release )  else (
cmake --build .
 cmake --build . --config Release
)
Error: could not load cache
Error: could not load cache

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim\external\rpclib\build>if ERRORLEVEL 1 goto :buildfailed

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim\external\rpclib\build>echo(

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim\external\rpclib\build>echo #### Build failed - see messages above.  1>&2
#### Build failed - see messages above.

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim\external\rpclib\build>chdir /d C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim\

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>exit /b 1

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>

RafaGuil, 2024-04-30:
Yes, I have the rpclib repo:
[image]

Also I removed AirSim/external/rpclib/build folder and then I update submodules, but when I do git submodule update --init --recursive, nothing happens in the console.

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>git submodule update --init --recursive

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>

wouter-heerwegh, 2024-04-30:
Have you installed these components for visual studio?
[image]

I did an installation from scratch, and it just worked. Does launching the developer console in administrator mode a difference?

RafaGuil, 2024-04-30:
Yes, I have installed everything:
[image]
[image]
[image]
[image]

Also I try to run it in administrator mode and there is no difference.

wouter-heerwegh, 2024-04-30:
Have you also installed the windows 11 SDK?

Something you can also still try is going inside the rpclib directory and running `git clean -xdf` and `git checkout .`

RafaGuil, 2024-04-30:
Yes, but it's not the 10.0.18.362.0 version (because it only exist this version)
[image]

About the git clean -xdf and git checkout . , it fails again:
C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>cd external

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim\external>cd rpclib

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim\external\rpclib>git clean -xdf
Removing build/

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim\external\rpclib>git checkout .
Updated 0 paths from the index

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim\external\rpclib>cd ../..

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>build.cmd

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>REM //---------- set up variable ----------

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>setlocal

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>set ROOT_DIR=C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim\

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>REM // Check command line arguments

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>set "noFullPolyCar="

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>set "buildMode="

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>REM //check VS version

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>if "16.0" == "" (
echo(
 echo oh oh... You need to run this command from x64 Native Tools Command Prompt for VS 2019.
 goto :buildfailed_nomsg
)

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>if "16.0" LSS "16.0" (
echo(
 echo Hello there! We just upgraded AirSim to Unreal Engine 4.24 and Visual Studio 2019.
 echo Here are few easy steps for upgrade so everything is new and shiny:
 echo https://github.com/Microsoft/AirSim/blob/master/docs/unreal_upgrade.md
 goto :buildfailed_nomsg
)

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>if "" == "" goto noargs

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>chdir /d C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim\

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>REM //---------- Check cmake version ----------

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>CALL check_cmake.bat

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>REM //---------- set up variable ----------

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>setlocal

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>set ROOT_DIR=C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim\

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>set cmake_minversion_minmaj="   3.   9"

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>set "cmake_version=    .    "

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>where /q cmake

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>if 0 EQU 0 (
for /F "tokens=3" %a in ('cmake --version | find "cmake version"') do set cmake_version=%a
 if "    .    " == "" (
echo Unable to get version of cmake.  1>&2
[... 107 lines cut ...]
                               = Generates Eclipse CDT 4.0 project files.
  Eclipse CDT4 - MinGW Makefiles
                               = Generates Eclipse CDT 4.0 project files.
  Eclipse CDT4 - Ninja         = Generates Eclipse CDT 4.0 project files.
  Eclipse CDT4 - Unix Makefiles= Generates Eclipse CDT 4.0 project files.

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim\external\rpclib\build>if "" == "--Debug" (cmake --build . --config Debug )  else if "" == "--Release" (cmake --build . --config Release )  else (
cmake --build .
 cmake --build . --config Release
)
Error: could not load cache
Error: could not load cache

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim\external\rpclib\build>if ERRORLEVEL 1 goto :buildfailed

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim\external\rpclib\build>echo(

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim\external\rpclib\build>echo #### Build failed - see messages above.  1>&2
#### Build failed - see messages above.

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim\external\rpclib\build>chdir /d C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim\

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>exit /b 1

C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim>

wouter-heerwegh, 2024-04-30:
You seem to have a different cmake version (3.9), mine is 3.20. Could you run `git clean -xdf` and `git checkout .` again, but on the AirSim folder?

In your log I see
```
 CMake Error: Could not create named generator Visual Studio 16 2019
```

On my end I see the following:
```
-- Selecting Windows SDK version 10.0.22621.0 to target Windows 10.0.22631.
```

closed 2024-04-30 by wouter-heerwegh