FS-Driverless/Formula-Student-Driverless-Simulator #369: Packaging from windows to ubuntu does not work

RafaGuil, 2024-05-09:
Hi. I've packaged the simulator for Windows and I didn't have any problem. But when I packaged it for Linux I haved this error:
UATHelper: Empaquetado (Linux): ********** BUILD COMMAND STARTED **********
UATHelper: Empaquetado (Linux): Running: C:\Users\Rafa Guil Valero\UE_4.27\Engine\Binaries\DotNET\UnrealBuildTool.exe Blocks Linux Development -Project="C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\UE4Project\FSOnline.uproject"  "C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\UE4Project\FSOnline.uproject" -NoUBTMakefiles  -r
emoteini="C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\UE4Project" -skipdeploy -distribution -Manifest="C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\UE4Project\Intermediate\Build\Manifest.xml" -NoHotReload -log="C:\Users\Rafa Guil Valero\AppData\Roaming\Unreal Engine\AutomationTool\Logs\C+Users+Rafa+Guil+Valero+UE_4.27\UBT
-Blocks-Linux-Development.txt"
UATHelper: Empaquetado (Linux):   Using 'git status' to determine working set for adaptive non-unity build (C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator).
UATHelper: Empaquetado (Linux):   WARNING: Library 'C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\UE4Project\Plugins\AirSim\Source\AirLib\deps\rpclib\lib\librpc.a' was not resolvable to a file when used in Module 'AirSim', assuming it is a filename and will search library paths for it. This is slow and dependency checking will not work for i
t. Please update reference to be fully qualified alternatively use PublicSystemLibraryPaths if you do intended to use this slow path to suppress this warning.
LogSlate: Took 0.000112 seconds to synchronously load lazily loaded font '../../../Engine/Content/Slate/Fonts/DroidSansMono.ttf' (77K)
UATHelper: Empaquetado (Linux):   Waiting for 'git status' command to complete
UATHelper: Empaquetado (Linux):   ------- Build details --------
UATHelper: Empaquetado (Linux):   Using toolchain located at 'C:/UnrealToolchains/v19_clang-11.0.1-centos7/x86_64-unknown-linux-gnu'.
UATHelper: Empaquetado (Linux):   Using clang (C:/UnrealToolchains/v19_clang-11.0.1-centos7/x86_64-unknown-linux-gnu\bin\clang++.exe) version '11.0.1' (string), 11 (major), 0 (minor), 1 (patch)
UATHelper: Empaquetado (Linux):   Using bundled libc++ standard C++ library.
UATHelper: Empaquetado (Linux):   Using lld linker
UATHelper: Empaquetado (Linux):   Using llvm-ar : C:/UnrealToolchains/v19_clang-11.0.1-centos7/x86_64-unknown-linux-gnu\bin/llvm-ar.exe
UATHelper: Empaquetado (Linux):   Using old way to relink circularly dependent libraries (with a FixDeps step).
UATHelper: Empaquetado (Linux):   ------------------------------
UATHelper: Empaquetado (Linux):   Writing manifest to C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\UE4Project\Intermediate\Build\Manifest.xml
UATHelper: Empaquetado (Linux):   Building Blocks...
UATHelper: Empaquetado (Linux):   Building 2 actions with 20 processes...
UATHelper: Empaquetado (Linux):     [1/2] Blocks
UATHelper: Empaquetado (Linux):     ld.lld: error: cannot open C:/Users/Rafa Guil Valero/Formula-Student-Driverless-Simulator/UE4Project/Plugins/AirSim/Source/AirLib/deps/rpclib/lib/librpc.a: no such file or directory
UATHelper: Empaquetado (Linux):     clang++: error: linker command failed with exit code 1 (use -v to see invocation)
UATHelper: Empaquetado (Linux):     ld.lld: error: cannot open C:/Users/Rafa Guil Valero/Formula-Student-Driverless-Simulator/UE4Project/Plugins/AirSim/Source/AirLib/deps/rpclib/lib/librpc.a: no such file or directory
UATHelper: Empaquetado (Linux):     clang++: error: linker command failed with exit code 1 (use -v to see invocation)
UATHelper: Empaquetado (Linux):     ld.lld: error: cannot open C:/Users/Rafa Guil Valero/Formula-Student-Driverless-Simulator/UE4Project/Plugins/AirSim/Source/AirLib/deps/rpclib/lib/librpc.a: no such file or directory
UATHelper: Empaquetado (Linux):     clang++: error: linker command failed with exit code 1 (use -v to see invocation)
UATHelper: Empaquetado (Linux):     ld.lld: error: cannot open C:/Users/Rafa Guil Valero/Formula-Student-Driverless-Simulator/UE4Project/Plugins/AirSim/Source/AirLib/deps/rpclib/lib/librpc.a: no such file or directory
UATHelper: Empaquetado (Linux):     clang++: error: linker command failed with exit code 1 (use -v to see invocation)
UATHelper: Empaquetado (Linux):     ld.lld: error: cannot open C:/Users/Rafa Guil Valero/Formula-Student-Driverless-Simulator/UE4Project/Plugins/AirSim/Source/AirLib/deps/rpclib/lib/librpc.a: no such file or directory
UATHelper: Empaquetado (Linux):     clang++: error: linker command failed with exit code 1 (use -v to see invocation)
UATHelper: Empaquetado (Linux):     ld.lld: error: cannot open C:/Users/Rafa Guil Valero/Formula-Student-Driverless-Simulator/UE4Project/Plugins/AirSim/Source/AirLib/deps/rpclib/lib/librpc.a: no such file or directory
UATHelper: Empaquetado (Linux):     clang++: error: linker command failed with exit code 1 (use -v to see invocation)
UATHelper: Empaquetado (Linux):     ld.lld: error: cannot open C:/Users/Rafa Guil Valero/Formula-Student-Driverless-Simulator/UE4Project/Plugins/AirSim/Source/AirLib/deps/rpclib/lib/librpc.a: no such file or directory
UATHelper: Empaquetado (Linux):     clang++: error: linker command failed with exit code 1 (use -v to see invocation)
UATHelper: Empaquetado (Linux):     ld.lld: error: cannot open C:/Users/Rafa Guil Valero/Formula-Student-Driverless-Simulator/UE4Project/Plugins/AirSim/Source/AirLib/deps/rpclib/lib/librpc.a: no such file or directory
UATHelper: Empaquetado (Linux):     clang++: error: linker command failed with exit code 1 (use -v to see invocation)
UATHelper: Empaquetado (Linux):     ld.lld: error: cannot open C:/Users/Rafa Guil Valero/Formula-Student-Driverless-Simulator/UE4Project/Plugins/AirSim/Source/AirLib/deps/rpclib/lib/librpc.a: no such file or directory
UATHelper: Empaquetado (Linux):     clang++: error: linker command failed with exit code 1 (use -v to see invocation)
UATHelper: Empaquetado (Linux):     ld.lld: error: cannot open C:/Users/Rafa Guil Valero/Formula-Student-Driverless-Simulator/UE4Project/Plugins/AirSim/Source/AirLib/deps/rpclib/lib/librpc.a: no such file or directory
UATHelper: Empaquetado (Linux):     clang++: error: linker command failed with exit code 1 (use -v to see invocation)
UATHelper: Empaquetado (Linux):     Failed to link C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\UE4Project\Binaries\Linux\Blocks after 10 retries
UATHelper: Empaquetado (Linux): Took 3,7330101s to run UnrealBuildTool.exe, ExitCode=6
UATHelper: Empaquetado (Linux): UnrealBuildTool failed. See log for more details. (C:\Users\Rafa Guil Valero\AppData\Roaming\Unreal Engine\AutomationTool\Logs\C+Users+Rafa+Guil+Valero+UE_4.27\UBT-Blocks-Linux-Development.txt)
UATHelper: Empaquetado (Linux): AutomationTool exiting with ExitCode=6 (6)
UATHelper: Empaquetado (Linux): BUILD FAILED

What can I do?

RafaGuil, 2024-05-10:
The file is not there. 
[image]

wouter-heerwegh, 2024-05-10:
Hi @RafaGuil 

The library should be present in `external\rpclib\build\Debug` could you temporarily copy is from here to `Formula-Student-Driverless-Simulator/UE4Project/Plugins/AirSim/Source/AirLib/deps/rpclib/lib/librpc.a`.

I think I will have to modify the build.cmd script from AirSim

RafaGuil, 2024-05-10:
I can't find librpc.a in `C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\AirSim\external\rpclib\build\Debug`.
[image]

In fact, there is no file called `librpc.a` in the entire repository:
[image]

wouter-heerwegh, 2024-05-10:
You have built AirSim, right? Because it should be built for AirSim.

RafaGuil, 2024-05-11:
Yes, I rebuilt it and it still doesn't work. 
[image]

That's what build.cmd says:
[building.txt](https://github.com/FS-Driverless/Formula-Student-Driverless-Simulator/files/15282469/building.txt)

wouter-heerwegh, 2024-05-11:
Hi @RafaGuil,

I just compiled for Windows without issue (Windows 11, Visual Studio v2019, Unreal Engine 4.27.2). It seems that the rpc library is built in Windows as `rpc.lib`.

Are you building for Windows aswell? Or for Linux?

RafaGuil, 2024-05-11:
I'm building for Linux. I have no problem for Windows.

wouter-heerwegh, 2024-05-11:
Unreal Engine seems to try to rebuild the AirSim plugin, and is trying to find `librpc.a`

If you want a quick and dirty solution, you could compile in WSL or on your linux system and copy the needed library files over. Important to note is that you will have to change the used clang version from v12 to v8, in order to build `librpc.a` without the  `pthread_cond_clockwait` symbol.

I quickly used WSL (Ubuntu 20.04), compiled AirSim, then opened the Unreal Editor, rebuild the plugin, and packaged for Linux succesfully.

I know that this is a terrible workflow, but I'll need some time to look into this. Feel free to let me know if you struggle with something.

RafaGuil, 2024-05-11:
I've copy `librpc.a` in `C:\Users\Rafa Guil Valero\Formula-Student-Driverless-Simulator\UE4Project\Plugins\AirSim\Source\AirLib\deps\rpclib\lib` from my Ubuntu 20.04 builded repo, and I have this error with -v19 clang-11.0.1-based:
[image]
then I tried with -v15 clang-8.0.1-based and it fails again:
[image]
Finally I tried again with -v8 clang-3.9.0-based In case I had misunderstood you, but it still fails.
[image]

I also tried to compile setup.sh and build.sh with clang8 on WSL (Ubuntu 20.04), but build.sh fails:
[image]

wouter-heerwegh, 2024-05-12:
Hi @RafaGuil,

Unreal Engine is complaining about the `pthread_cond_clockwait` symbol, so the library was probably not rebuild after changing both the setup and build script to use clang-8. It's best to remove the build_debug folder before rebuilding, and you can check if the symbol is still present by checking the library with the following command:
```
nm librpc.a | grep pthread_cond
```
This command should not return pthread_cond_clockwait when building with clang 8.

RafaGuil, 2024-05-12:
Hi, I rebuilded the airsim with clang-11 (Unreal Engine doesn'work with clang-8) and the pthread_cond_clockwait symbol doesn't appear any more:
[image]
Then, I went to Unreal Engine, I packaged for Linux and it still doesn't work. It's a different error (a very big error), but it doesn't work:
[image]

wouter-heerwegh, 2024-05-12:
It seems to complain about standard symbols. Unreal engine is building with clang v11, but the rpclib can still be built with clang 8. Make sure you have the following packages installed for your clang version:
```
clang-12 clang++-12 libc++-12-dev libc++abi-12-dev
```

RafaGuil, 2024-05-13:
Hi @wouter-heerwegh 

I was investiganting and I realized that I was using the wrong `librpc.a` (the one from ros_bridge) that is the reason for the error. Now I am using the correct one and the error persist, but it is because of `pthread_cond_clockwait`. How can I remove `pthread_cond_clockwait`? I tried to eliminate it but I couldn't do it.
[image]

RafaGuil, 2024-05-13:
Nup, it doesn't work:
[image]

wouter-heerwegh, 2024-05-13:
Looking at the prompt above, it seems that the build script did not finish completely successful.
The script should show
```
==================================================================
 AirSim libraries are built and installed.
==================================================================
```

RafaGuil, 2024-05-13:
I'll try to reinstall the entire repository. It seems that there are some files that I was able to delete by mistake, it is inexplicable that it works for you and it doesn't work for me, having everything the same. I will tell you tomorrow.

RafaGuil, 2024-05-14:
Hi @wouter-heerwegh,

I reinstalled the repo and also I changed to WSL Ubuntu 18.04 like you say in software install instructions. I run setup.sh and now it is building CMake, but it's taking too long. Is it normal? It's the first time that setup.sh takes me this long.

wouter-heerwegh, 2024-05-14:
You could manually install cmake, using the apt repository. Otherwise it will build cmake from scratch, which might take some time.

RafaGuil, 2024-05-14:
Build.sh failed again...
[building.txt](https://github.com/FS-Driverless/Formula-Student-Driverless-Simulator/files/15307336/building.txt)

wouter-heerwegh, 2024-05-14:
Not sure why the script is failing here, because the clean.sh file is definitely present in the repo

RafaGuil, 2024-05-14:
Clean.sh is related to Unreal Engine, maybe should I install Unreal Engine for Ubuntu? Or somehow import the Windows Unreal Engine to WSL?

wouter-heerwegh, 2024-05-14:
No, you don't need to install Unreal Engine. 

I think I know what is going on. On Windows it will append lines in a file with `^M` (see [this post](https://stackoverflow.com/questions/64749/m-character-at-end-of-lines)). Linux can not handle these characters. Can you run `git checkout . -- UE4Project/clean.sh`? This should reset the file to the original one.

It is possible that some extension in vscode or another IDE is changing these end of line characters.

RafaGuil, 2024-05-14:
One fewer problem :), I was doing dos2unix on setup.sh and build.sh, but I didn't know that I have to do it with clean.sh too...
[image]

...but Unreal Engine doesn't work:
[image]

wouter-heerwegh, 2024-05-14:
Does this happen when opening the unreal engine project?

If so, build AirSim for windows first, open the unreal engine project, close it, build AirSim in WSL, open the UE project and package for linux

RafaGuil, 2024-05-14:
FINALLY!!!
[image]

One last question, where is the settings.json file?

wouter-heerwegh, 2024-05-14:
It is stored in the root of the repo, you should be able to just copy it into the built directory. There is still a bug I need to fix in order to fix the camera_launcher however. So unless you use camera's all should be fine.

Would you perhaps be willing to share the executable/changes to the handling? The current handling is not really amazing, so any improvement is welcome.

RafaGuil, 2024-05-14:
Okey perfect!

This afternoon I'm gonna make a pdf with all the changes and then I will publish it right here. Thank you for everything!!!

closed 2024-05-14 by RafaGuil

RafaGuil, 2024-05-14:
Here you go:
[steep_by_steep.txt](https://github.com/FS-Driverless/Formula-Student-Driverless-Simulator/files/15311439/steep_by_steep.txt)

If you need anything else, tell me!

wouter-heerwegh, 2024-05-14:
Hi @RafaGuil ,

Thanks for the quick recap of how you got it to work. You said you were making some changes on the dynamics of the car. Would you be willing to share some of the changes you made?

The car at the moment does not react all that nice imo, so any feedback is appreciated.

RafaGuil, 2024-05-16:
Here you go! This setup could be better, but I preferred to make a realistic one.
[Dynamic data.txt](https://github.com/FS-Driverless/Formula-Student-Driverless-Simulator/files/15338580/Dynamic.data.txt)

wouter-heerwegh, 2024-05-16:
Thanks @RafaGuil, I'll try them out