FS-Driverless/Formula-Student-Driverless-Simulator #372: Problem with installation FSDS from source in ubuntu 22.04

RoyAmoyal, 2024-10-16:
Hey,

I am opening the issue because we can't compile from source the project on our Ubuntu 22.04. The solution in #365 will not work here because we can't use clang 8 in Ubuntu 22.04

[image]

Hopefully it will be solved soon, our Formula Student group really want to use that simulator. 

Thanks!

wouter-heerwegh, 2024-10-16:
Hi @RoyAmoyal 

You can perfectly use Ubuntu 22.04 and ROS2. The only issue for now is that if you want to edit files in unreal engine, you can build the Airsim library in an Ubuntu 18.04 docker container. After building the library (with the setup and build script), you can close the container and you should be able to launch the unreal project.

RoyAmoyal, 2024-10-17:
> Hi @RoyAmoyal
> 
> You can perfectly use Ubuntu 22.04 and ROS2. The only issue for now is that if you want to edit files in unreal engine, you can build the Airsim library in an Ubuntu 18.04 docker container. After building the library (with the setup and build script), you can close the container and you should be able to launch the unreal project.
> 
> I know it's not an ideal scenario, but it should work for now. I'll try to look again to use [the cross compile toolchain from Unreal Engine](https://dev.epicgames.com/documentation/en-us/unreal-engine/linux-development-requirements-for-unreal-engine?application_version=5.5#versionhistory)

So If I compile the project from source on the docker with ubuntu 18.04, how do I use it in my ubuntu 22.04? 

and which docker should I use? Thanks!

wouter-heerwegh, 2024-10-17:
Hi @RoyAmoyal,

You don't have to build the Unreal Engine project inside the container, only the AirSim library. So only [this step](https://fs-driverless.github.io/Formula-Student-Driverless-Simulator/v2.2.0/getting-started/#2-compiling-the-airsim-plugin) (step 2 in the installation). All other steps are done on the host system.

Make sure to mount the repo as a volume in your docker container, that way you can just build the project on your host after.

RoyAmoyal, 2024-10-17:
> Hi @RoyAmoyal,
>
> You don't have to build the Unreal Engine project inside the container, only the AirSim library. So only [this [...]

I tried that:

My dockerfile:
```
# Use Ubuntu 18.04 as the base image
FROM ubuntu:18.04

# Set environment variables
ENV DEBIAN_FRONTEND=noninteractive

# Update and install necessary packages
RUN apt-get update && apt-get install -y \
    git \
    build-essential \
    cmake \
    libboost-all-dev \
    libasio-dev \
    libtinyxml2-dev \
    libcurl4-openssl-dev \
    libglfw3-dev \
    libglew-dev \
    libpng-dev \
    libjpeg-dev \
    libopencv-dev \
    libtiff-dev \
    python3 \
    python3-pip \
    python3-dev \
    unzip \
    wget \              
    lsb-release \      
    && apt-get clean \
    && rm -rf /var/lib/apt/lists/*

# Install required packages
RUN apt-get update && \
    apt-get install -y git build-essential wget software-properties-common gnupg rsync && \
    apt-get clean

# Create a directory for the project
WORKDIR /home/airsim

# Clone the Formula Student Driverless Simulator repository
RUN git clone --recurse-submodules https://github.com/FS-Driverless/Formula-Student-Driverless-Simulator.git

# Set the working directory to the AirSim folder
WORKDIR /home/airsim/Formula-Student-Driverless-Simulator/AirSim

# Modify the setup.sh script to remove 'sudo'
RUN sed -i 's/sudo //g' setup.sh

# Make scripts executable and run them
RUN chmod +x setup.sh build.sh && \
    ./setup.sh && \
    ./build.sh

# Set the entry point
ENTRYPOINT ["/bin/bash"]
```

Then I run: 

`sudo docker run --rm -it -v /home/roy/Formula-Student-Driverless-Simulator:/home/airsim/Formula-Student-Driverless-Simulator fs_driverless_simulator`

and when I tried to build with the next command:
`~/computer_graphics/UnrealEngine/Engine/Binaries/ThirdParty/Mono/Linux/bin/mono ~/computer_graphics/UnrealEngine/Engine/Binaries/DotNET/UnrealBuildTool.exe Development Linux -Project=/home/roy/Formula-Student-Driverless-Simulator/UE4Project/FSOnline.uproject -TargetType=Editor -Progress
`

I got the error:
```
...
@progress 'Generating code...' 100%
@progress pop
Building BlocksEditor...
Performing 3 actions (20 in parallel)
@progress 'Compiling C++ source code...' 0%
[1/3] Link (lld) libUE4Editor-AirSim.so
ld.lld: error: undefined symbol: pthread_cond_clockwait
>>> referenced by __mutex_base:508 (/usr/lib/llvm-12/bin/../include/c++/v1/__mutex_base:508)
>>>               client.cc.o:(std::__1::condition_variable::__do_timed_wait(std::__1::unique_lock<std::__1::mutex>&, std::__1::chrono::time_point<std::__1::chrono::steady_clock, std::__1::chrono::duration<long long, std::__1::ratio<1l, 1000000000l> > >)) in archive /home/roy/Formula-Student-Driverless-Simulator/UE4Project 4.27 - 3/Plugins/AirSim/Source/AirLib/deps/rpclib/lib/librpc.a
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
@progress 'Compiling C++ source code...' 25%
@progress 'Compiling C++ source code...' 50%
```

Any idea how to continue? Thanks!!

wouter-heerwegh, 2024-10-17:
Hi @RoyAmoyal,

Not sure if you will have to clone again, since you are connecting it to a volume outside your container that already contains the repo, but should be fine. I see in the logs you posted the script is using clang 12 (and llvm12). Could you try again with all references in the setup and build scripts converted to clang 8 and llvm8 instead?

RoyAmoyal, 2024-10-19:
> Hi @RoyAmoyal,
>
> Not sure if you will have to clone again, since you are connecting it to a volume outside your container that already contains the repo, but should be fine. I see in the logs you [...]

I managed to compile it using the docker, Thanks!

Maybe I will contribute a documentation about how to install it on Ubuntu 22.04 (and 20.04). Is it still possible to contribute to the project?

wouter-heerwegh, 2024-10-19:
Sure, that would be great thanks

wouter-heerwegh, 2024-10-24:
Closing because of changes done in #374

closed 2024-10-24 by wouter-heerwegh