FS-Driverless/Formula-Student-Driverless-Simulator #373: Problem with packging and shipping the project in ubuntu 22.04

RoyAmoyal, 2024-10-19:
Hey,

I used docker (Ubuntu 20/18) with clang 8 to build the Airsim inside the docker and mounting it to my host  with Ubuntu 22.04.

It worked, and I can edit the project with Unreal with my Ubuntu 22.04.

Now when I try to build and compile the project I get erros like:

``` 
UnrealEngine/Engine/Source/Runtime/Core/Public/Templates/TypeHash.h:102:27: [0m[0;1;31merror: [0m[1munknown type name 'uint64_t'[0m
PackagingResults: Error: [0m[1munknown type name 'uint64_t'[0m

PackagingResults: Error: [0m[1mno member named 'GetCachedContainer' in 'FCachedPropertyPath'[0m
UATHelper: Packaging (Linux):   [1m/home/myuser/computer_graphics/UnrealEngine/Engine/Source/Runtime/PropertyPath/Public/PropertyPathHelpers.h:456:39: [0m[0;1;31merror: [0m[1mno member named 'GetCachedContainer' in 'FCachedPropertyPath'[0m

PackagingResults: Error: [0m[1mno member named 'GetCachedContainer' in 'FCachedPropertyPath'[0m

UATHelper: Packaging (Linux):   [683/685] Compile Module.DynamicMesh.1_of_3.cpp
UATHelper: Packaging (Linux): Took 655.860589s to run mono, ExitCode=6
UATHelper: Packaging (Linux): UnrealBuildTool failed. See log for more details. (/home/roy/Library/Logs/Unreal Engine/LocalBuildLogs/UBT-Blocks-Linux-Shipping_2.txt)
UATHelper: Packaging (Linux): AutomationTool exiting with ExitCode=6 (6)
UATHelper: Packaging (Linux): RunUAT ERROR: AutomationTool was unable to run successfully.
PackagingResults: Error: AutomationTool was unable to run successfully.
PackagingResults: Error: Unknown Error
```
ChatGPT suggests that some of the errors related to the Clang version (probabily I used Airsim do compile inside the docker with clang 8, but the rest of the project, e.g. Unreal I compiled on my host with clang12)

What can I do to solve that? Do I need to use the docker again (and building the Unreal their too) and use it to compile inside the docker through commandline with Unreal?

Any other idea?

T!hanks

wouter-heerwegh, 2024-10-19:
Hi @RoyAmoyal,

Normally the project is packaged using the toolchain from unreal engine which uses clang 11. It should not be an issue, otherwise we would have seen this on Ubuntu 18.04 as well. `FCachedPropertyPath` is really a structure from Unreal Engine, so you could try to create an issue on their repo. The chance they will respond might be small, because it's for unreal engine 4.27.

If nothing works, you can try to use #370, since it uses Unreal Engine 5.3. We should merge it, but that means we will lose native ROS1 support, since you would need Ubuntu 22.04 to use Unreal Engine 5.3. Noetic is going EOL in May next year, so we might just merge and drop support.

RoyAmoyal, 2024-10-20:
After experimenting with it for a while, I've found that packaging in Development mode works correctly, but Shipping mode does not. 

It seems this issue may be related to a compatibility problem between the current version of Airsim and the Unreal Engine version.

Could you verify if the latest repository and installation of this project can be packaged in Shipping mode? Just to be sure.

btw I have added '#include <stdint.h>' to '/home/myuser/UnrealEngine/Engine/Source/Runtime/Core/Public/Templates/TypeHash.h' to fix the errors for Development mode packaging.

RoyAmoyal, 2024-10-20:
I managed to solve the problem. 

I comment out the lines 354 and 456 
`//check(InContainer == InPropertyPath.GetCachedContainer());` 

and it finished the shipping.

I need another help. After I shipped the project, I got 

```/home/roy/Desktop/tesk_pack2/LinuxNoEditor/FSOnline/Binaries/Linux/Blocks-Linux-Shipping
/home/roy/Desktop/tesk_pack2/LinuxNoEditor/FSOnline/Content/Paks/FSOnline-LinuxNoEditor.pak
/home/roy/Desktop/tesk_pack2/LinuxNoEditor/Blocks.sh
/home/roy/Desktop/tesk_pack2/LinuxNoEditor/Manifest_NonUFSFiles_Linux.txt
```

Thanks for all your help, @wouter-heerwegh! I'll soon be delivering the project to my Simulator Team at BGRacing Formula Student at our university. We're planning to start contributing to the project, as there are some functionalities we need that aren't currently available. I'll check in with you about these soon.

closed 2024-10-20 by RoyAmoyal

wouter-heerwegh, 2024-10-20:
Hi @RoyAmoyal,

Glad you were able to fix it. In which files did you comment those lines, so we know for future reference?

Development should be fine, it's also what we use for the packaged executables in the releases. I know I packaged them once as release builds, but can't remember why I didn't any more afterwards. Might be that some functionality was impacted.

RoyAmoyal, 2024-10-20:
wss

> Hi @RoyAmoyal,
>
> Glad you were able to fix it. In which files did you comment those lines, so we know for future reference?
>
> Development should be fine, it's also what we use for the packaged [...]

Yeah, sorry, I forgot to write the path, I edited the comment with the file path.

The shipping packaging project is very laggy, something is not optimized correctly. The development packaged project works great and runs smoothly!

So that's great, I am on the right track.