FS-Driverless/Formula-Student-Driverless-Simulator #379: Problem with 3d import car model behavior

ShaharHalevi, 2024-12-17:
Hey,
We are working on our custom car model, following the guide in the documentation.
When we run the simulation the car starts flipping uncontrollably we attached a video below.
Would really appreciate some help with this issue
Thanks!

https://github.com/user-attachments/assets/c4e2da47-ca1b-423a-a7ca-eaa0d2e4eae7

wouter-heerwegh, 2024-12-19:
Hi @ShaharHalevi,

Is it me or does your skeleton under ground? It looks a bit strange. I never really created a vehicle in Unreal, so I'm not immediately sure what the issue could be.

jachaussloef, 2025-02-20:
Sometimes you need to check your CoG position which is the origin point of your chassis object. Also pay attention to the unit of your exported fbx model. Try to make sure it's centimeter which is UE's default unit, and in UE check the imported model skeleton that the chassis scale is (1,1,1) and wheel position is in cm unit or the wheel position will be very strange (if it's meter for example all four wheels would be almost at the COG of the car even if the displayed position looks right). If any of the mentioned above is not satisfied, you might need to calculate and manually input the offsets of your wheels and/or CoG.